RobustFieldAutonomyLab/LeGO-LOAM #1: still suffering from Z drift, and loop closure detection failed

anjiss, 2018-07-11:
I've test my rosbag but z drift still exists, so after one loop, baselink might drift far enough that ICP can't handle it.

Is this possible due to the tilt between Velodyne and baselink?

anjiss, 2018-07-12:
https://drive.google.com/open?id=1fRhBjucR5oHgvIsV7QS_M6wveND5JKsn

I forgot to mention you that, the Lidar data im using is published circle by circle, without any corrections from motion, and for about 1/4 horizontal angles are occluded. 

Hope for your kind suggestions, many thanks!

TixiaoShan, 2018-07-12:
I observed no drift without loop closure after make two changes. Since your velodyne is not level and I don't know the frame of your imu, using imu data might make the result much worse. So I am not using imu data in your bag at all.

extern const int groundScanInd = 16; (in utility.h)
if (abs(angle - sensorMountAngle) <= 20) (in imageProjection.cpp)

You can make some tweaks to the "20" above and find the best value for your data.

[image]

closed 2018-07-16 by TixiaoShan