RobustFieldAutonomyLab/LeGO-LOAM #4: About the leGO-LOAM paper

Jingwen1995, 2018-07-23:
Hi! I'm really impressed by the speed and effect brought by LeGO-LOAM after playing my own rosbag.
Therefore, I really want to have some further understanding about how leGO-LOAM works. But I was unable to find the paper "LeGO-LOAM: Lightweight and Ground-Optimized Lidar Odometry and Mapping on Variable Terrain" online. 
So could you please upload the paper or provide some information about where I can find it?
Thanks a lot and have a nice day!

TixiaoShan, 2018-07-23:
@Jingwen1995 
Thanks for trying it out! The final draft will be ready by the end this week. It will be uploaded to the site below soon. I will mention you again once it is uploaded.
http://personal.stevens.edu/~benglot/publications.html

TixiaoShan, 2018-07-30:
@Jingwen1995 
https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/master/IROS_2018_Tixiao.pdf

closed 2018-07-30 by TixiaoShan