RobustFieldAutonomyLab/LeGO-LOAM #9: Applicability for Quadcopter?

AmineElhafsi, 2018-08-01:
Hello,

Would LeGO-LOAM be suitable to use for aerial vehicle localization/pose estimation in indoor, gps-denied settings (say, using a VLP16)? I noticed from the readme that the requirement is that the ground plane always needs to be in view. Can this requirement be relaxed in any way? 

Thanks!

AmineElhafsi, 2018-08-04:
Unfortunately at the moment I don't, but will get back to you if I record any.

TixiaoShan, 2018-08-10:
Close the topic for now. Reopen once with required data.

closed 2018-08-10 by TixiaoShan

droneboxid, 2018-11-09:
hi  @TixiaoShan 

i have try with RSLIDAR16 with change 

`subLaserCloud = nh.subscribe<sensor_msgs::PointCloud2>("/rslidar_points", 1, &ImageProjection::cloudHandler, this);`

but error

[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[imageProjection-5] process has died [pid 776, exit code -11, 

can you let me know ? also in my case need to combine with dji_sdk/imu 
can you try my bag file from drone with top down lidar position 

[https://drive.google.com/file/d/1ab0xSVm7J2RuqiGovNFjLuf4dzFsGIqi/view?usp=sharing](url)

thanks