RobustFieldAutonomyLab/LeGO-LOAM #10: Sudden rotation occurs during mapping

Oofs, 2018-08-15:
Hi Shan,

I  have tested your algorithm with my own datasets (VLP-16, no imu data), in dataset1, map is perfect with little drift (pic1, pic2). But in dataset2, it occurs a sudden rotation in the built map(pic3, pic4). 
My lidar is mounted on the surface of handcart (1m height) which moves smoothly.
Do you have any idea about this ? Or do you have any suggestions about which parameters should be adjusted accordingly ?
[image]
[image]
[image]
[image]

Yours,
Claude

TixiaoShan, 2018-08-15:
What's the rotation rate of your VLP-16? It looks like it's rotating at 5Hz.

deliangye, 2018-08-15:
I run into a similar issue when I test with gates_oscillating_motion.bag from JI Zhang.

TixiaoShan, 2018-08-15:
@deliangye 
Yeah, I do remember people mention that the map would diverge even when the original LOAM is used.

Oofs, 2018-08-16:
Thanks for your  replies ! @TixiaoShan @deliangye 

My VLP-16 rate is 10hz(600RPM). I guess maybe it's the problem of the way recording bag file.
I use `roslaunch velodyne_pointcloud VLP16_points.launch pcap:="/.../"` to play the data while another thread is recording the pointcloud to the bag file `rosbag record /velodyne_points`.

How do you guys convert pcap to bag ?

TixiaoShan, 2018-08-16:
@Oofs 
When I run that package, there is a /velodyne_points topic I can subscribe to. I can use it for online mapping or recording a bag.

Oofs, 2018-08-16:
@TixiaoShan 
Thats exactly the way i did. But I found that when i use velodyne_pointcloud to play the .pcap files, it appears that the point_cloud is not read correctly and incomplete (some circular sector shadow in the pointcloud).
But when i use veloview to read pcap file, the pointcloud seems complete.
Have you ever experienced this ?
[image]
[image]

Oofs, 2018-08-16:
@TixiaoShan 
well, seems like a known bug when using velodyne_pointcloud to read pcap files.
Thanks anyway :)  https://github.com/ros-drivers/velodyne/issues/116

TixiaoShan, 2018-08-20:
Close the topic for now. Feel free to re-open with new data.

closed 2018-08-20 by TixiaoShan