RobustFieldAutonomyLab/LeGO-LOAM #13: problem with rviz?

shibowing, 2018-08-22:
hi,  I am really sorry to bother you.  However,  i happen to a weird problem. When I try to roslaunch the run.launch(your provided ) and the bag (your provided), the rviz will crash.  The error messages are listed as following: I am using ubuntu 16.04 with ros kinect version

rviz: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreNode.cpp:405: virtual void Ogre::Node::setPosition(const Ogre::Vector3&): Assertion `!pos.isNaN() && "Invalid vector supplied as parameter"' failed.
[rviz-2] process has died [pid 26222, exit code -6, cmd /opt/ros/kinetic/lib/rviz/rviz -d /home/machozhao/zhao_ws/src/LeGO-LOAM/LeGO-LOAM/launch/test.rviz __name:=rviz __log:=/home/machozhao/.ros/log/f88fc50a-a5bd-11e8-9edc-70f11c0dd839/rviz-2.log].
log file: /home/machozhao/.ros/log/f88fc50a-a5bd-11e8-9edc-70f11c0dd839/rviz-2*.log

 rviz: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:252: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
[rviz-2] process has died [pid 25433, exit code -6, cmd /opt/ros/kinetic/lib/rviz/rviz -d /home/machozhao/zhao_ws/src/LeGO-LOAM/LeGO-LOAM/launch/test2.rviz __name:=rviz __log:=/home/machozhao/.ros/log/8be59486-a5bb-11e8-9edc-70f11c0dd839/rviz-2.log].
log file: /home/machozhao/.ros/log/8be59486-a5bb-11e8-9edc-70f11c0dd839/rviz-2*.log

I guess the test.rviz may have some problems, so I use my configured rviz file. Unfortunately,  the rviz also crashes. The most weird thing is that there will be fine if i run other ros packages with rviz file.  Can you give me some suggestions?

related problems:https://github.com/ros-visualization/rviz/issues/808

TixiaoShan, 2018-08-22:
@shibowing 
Hmmm, never had this problem before. Did you try to run it from a different computer? I ran the package on at least 5 different computers (laptops, desktops) and encountered no problems.

closed 2018-08-24 by TixiaoShan

goldenminerlmg, 2018-09-26:
I also experienced the same problem. This may caused by using bad or incomplete dataset. Change a public dataset to try it again. By the way, when I test in an incomplete dataset the first time, it run well. This is strange.