RobustFieldAutonomyLab/LeGO-LOAM #14: Evaluation on kitti dataset and video production

goldenminerlmg, 2018-08-22:
Dear Author:
      Very nice work! 
      1. I am wonder how you evaluate lego-loam on kitti dataset. I used the tool(https://pypi.org/project/kitti2bag/1.5/) to convert the raw data into rosbag. when I run the rosbag, there are msgs lost. Is this caused by hardware resources limitations? I saw your paper has evaluate the algorithm on tx2, have you encounter such problems?
      2. The demo video is very interesting. How do you dynamically display trajectories to Google Maps in real time?  Have you used any video production software ? 
     Thanks very much!

TixiaoShan, 2018-08-22:
@goldenminerlmg 
1. We converted the point cloud data in KITTI dataset to a rosbag. To test KITTI dataset on Jetson, the point cloud data is downsampled to only 16 beams. I didn't observe message drops. Maybe it's the problem of your bags? Keep in mind that point cloud data in KITTI dataset is distortion free, LeGO-LOAM is not the ideal tool to perform accurate pose estimation since many points may never be projected onto the range image correctly (see previous issues). The result in the paper just shows the proof of concept of loop-closure.
2. I used Final Cut Pro X to add that panel.

goldenminerlmg, 2018-08-23:
Thanks very much!
So how did you downsampled the beams to 16 lines. Can I use the downsample method like 'voxelgrid' in PCL to directly operate the raw rosbag(64 beams). I saw your paper that the point must corresponding to the range image. It seams that only the raw points can be used, and we can't use the center of gravity of the voxel (Voxelgrid downsample) to approximate the raw point?

TixiaoShan, 2018-08-23:
@goldenminerlmg 
First, I still project the HDL-64E point cloud onto a 64*1800 range image. Then I only choose 16 lines from the total of 64 lines for LeGO-LOAM. For example, there are 1,2,3,4,5,6,7,...64 lines. I select 1,5,9,13, ... lines while deleting the rest.

goldenminerlmg, 2018-08-23:
Thank you for your careful answer！

closed 2018-08-23 by TixiaoShan