RobustFieldAutonomyLab/LeGO-LOAM #18: Drift found in localisation of LeGO-LOAM algorithm using rosbag captured from CARLA

praveenputhettu, 2018-09-05:


praveenputhettu, 2018-09-05:
The Lidar sensor used in capturing the point cloud is VLP-16

praveenputhettu, 2018-09-05:
Following is the trajectory i got before loop closure
[image]

praveenputhettu, 2018-09-05:
Following is the trajectory i got after loop closure
[image]

the loop closure happens only at the end of the path.
The actual path has no angled junctions as in the screen-shots.
i didn't use IMU here.
Can you identify which were the possible reasons of the error.

Thank You.

TixiaoShan, 2018-09-08:
Hi @praveenputhettu , sorry I don't have experience with using simulated point cloud in LeGO-LOAM. LeGO-LOAM is only designed for real lidar.

closed 2018-09-08 by TixiaoShan