RobustFieldAutonomyLab/LeGO-LOAM #20: I think the point cloud should remove the Nan point in the call back.

306327680, 2018-09-10:
Hello, I use the RoboSense lidar and when I run it the program is broken and finally, I find that the point cloud contains nan point, I think it should add a removeNaNFromPointCloud when point cloud comes in.

TixiaoShan, 2018-09-10:
Hi @306327680 , agreed. If your point cloud driver gives you Nan point, you should definitely add that function, which is used in the original LOAM code. Because my Velodyne driver does't give any Nan point in the cloud, I deleted that function.

closed 2018-09-10 by TixiaoShan

HappySamuel, 2019-05-31:
Hi @TixiaoShan 

How should i add the NAN point removal function into the code. As i use the Ouster OS1-16 lidar, there is NAN points generated.

Best,
Samuel

zorosmith, 2019-05-31:
you can use another ros node which subscribes to your raw point cloud firstly, removes NAN points and then publish the point cloud topic again.