RobustFieldAutonomyLab/LeGO-LOAM #21: The kitti raw datasets's IMU data for your algorithm

irobotDavinci, 2018-09-12:
Hi,I want to use the IMU data in kitti raw datasets to test lego ,just remap '/imu/data' to '/kitti/oxts/imu'
but it goes wrong when turn rotation on z axis.  For example：
[image]

And When not use IMU data，it will be ok:

[image]

The kitti raw datasets coordinate transform between velodyne and imu like this:
[image]

And in your datasets 2017-06-08-15-49-45_0.bag , the tf is like this:
[image]

Can help me what's wrong?

By the way, is there any doc about featureAssociate.cpp's detail. Thanks

irobotDavinci, 2018-09-12:
In addition, I use the kitti raw datasets is synced

zorosmith, 2019-01-13:
[image]

zorosmith, 2019-01-13:
[image]

zorosmith, 2019-01-13:
In the author's another dataset 2017-06-08-15-51-45_2.bag, the tf is shown in the following picture
[image]
This one is different from iroboDavinci's one, which one is correct?

closed 2019-05-30 by TixiaoShan

narutojxl, 2019-08-01:
Hi @irobotDavinci  @TixiaoShan , i also meet this problem. After convert  imu's orientation, acc and angular vel to imu' frame, which has the same origin with original imu frame but align with laser frame,  the problem not solved. I test with kitti_2011_10_03_drive_0027_synced  raw data.