RobustFieldAutonomyLab/LeGO-LOAM #24: Problems with VLP-16 dataset w/ IMU

ojura, 2018-09-19:
Hi! Would you have a moment to take a look at the dataset provided here?
https://ferhr-my.sharepoint.com/:f:/g/personal/jorsulic_fer_hr/EvwbIJ1tVipLqAJi1ECinDcBLLwKVF0Vki1m4zQ6ye0Jwg?e=TqqWCu

To run Lego-LOAM with the dataset:
```
rosrun imu_filter_madgwick imu_filter_node /imu/data_raw:=imu _publish_tf:=false _use_mag:=false _gain:=0.005
roslaunch lego_loam run.launch
rosbag play processed_type1_traderjoes.bag  points2_1_VLP16:=/velodyne_points --clock
```

The trajectory sometimes looks okay piecewise, but often has jerky moments. Also, in the second part of `processed_type1_traderjoes.bag`, the trajectory goes crazy, up in the air, with an arbitrary pitch and roll - it looks like the gravity measured by the IMU is ignored. As far as I can tell, the IMU is well aligned to the lidar (~< 0.5 degrees). Do you maybe know what could be the cause?

ksk0050, 2018-10-19:
Check 'extrinsic parameters' between Lidar and IMU.

ojura, 2018-11-17:
Hi @TixiaoShan, is there a chance you could take a look?

closed 2019-09-06 by stale[bot]