RobustFieldAutonomyLab/LeGO-LOAM #26: package did not work using velodyne VLP-16

chobabo, 2018-09-27:
I can compile the package and play sample bag files. But, package did not work using velodyne VLP-16.

$roscore

//http://wiki.ros.org/velodyne/Tutorials/Getting%20Started%20with%20the%20Velodyne%20VLP16
$roslaunch velodyne_pointcloud VLP16_points.launch

$roslaunch lego_loam run.launch

w4rlock999, 2018-10-16:
@chobabo I haven't try it yet on real time, but i have a bag file using of my own VLP16 and I could only run the lego-loam using my bag with:

$rosbag play the_name_of_the_bagfile.bag --clock --topic /velodyne_points

where the regular

$rosbag play the_name_of_the_bagfile.bag 

doesn't work, i dont know why

closed 2018-10-18 by chobabo

gkharish, 2019-05-03:
@chobabo , Have you changed the parameter /use_sim_time to false in run.launch? 
In order to make lego_loam work with the real velodyne, you must change the parameter /use_sim_time to false.