RobustFieldAutonomyLab/LeGO-LOAM #27: catkin_make error on ubuntu 16.04 + ROS kinetic

XuanboWang, 2018-09-28:
I try to make this package and get the following error:

[ 66%] Building CXX object LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o
In file included from /home/wangbx/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:36:0:
/usr/local/include/gtsam/geometry/Rot3.h: In static member function ‘static gtsam::Rot3 gtsam::Rot3::rodriguez(double, double, double)’:
/usr/local/include/gtsam/geometry/Rot3.h:197:51: error: no matching function for call to ‘gtsam::Rot3::rodriguez(Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >&)’
       { return rodriguez((Vector(3) << wx, wy, wz));}
                                                   ^
/usr/local/include/gtsam/geometry/Rot3.h:164:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::rodriguez(const Vector&, double)
     static Rot3 rodriguez(const Vector& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:164:17: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/gtsam/geometry/Rot3.h:172:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::rodriguez(const gtsam::Point3&, double)
     static Rot3 rodriguez(const Point3& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:172:17: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/gtsam/geometry/Rot3.h:180:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::rodriguez(const gtsam::Unit3&, double)
     static Rot3 rodriguez(const Unit3& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:180:17: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/gtsam/geometry/Rot3.h:187:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::rodriguez(const Vector&)
     static Rot3 rodriguez(const Vector& v);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:187:17: note:   no known conversion for argument 1 from ‘Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >’ to ‘const Vector& {aka const Eigen::Matrix<double, -1, 1>&}’
/usr/local/include/gtsam/geometry/Rot3.h:196:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::rodriguez(double, double, double)
     static Rot3 rodriguez(double wx, double wy, double wz)
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:196:17: note:   candidate expects 3 arguments, 1 provided
In file included from /home/wangbx/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:37:0:
/usr/local/include/gtsam/geometry/Pose3.h: In static member function ‘static gtsam::Matrix gtsam::Pose3::wedge(double, double, double, double, double, double)’:
/usr/local/include/gtsam/geometry/Pose3.h:226:28: error: could not convert ‘(&(&(&(&(&(&(&(&(&(&(&(&(&(& Eigen::DenseBase<Derived>::operator<<(const Scalar&) [with Derived = Eigen::Matrix<double, -1, -1>; Eigen::DenseBase<Derived>::Scalar = double](0.0).Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((- wz)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(wy))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(vx))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(wz))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(0.0))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((- wx)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(vy))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((- wy)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(wx))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(0.0))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(vz))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(0.0))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(0.0))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(0.0))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >(0.0)’ from ‘Eigen::CommaInitializer<Eigen::Matrix<double, -1, -1> >’ to ‘gtsam::Matrix {aka Eigen::Matrix<double, -1, -1>}’
           0.,  0.,  0.,  0.);
                            ^
In file included from /usr/local/include/gtsam/nonlinear/NonlinearFactorGraph.h:24:0,
                 from /home/wangbx/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:40:
/usr/local/include/gtsam/geometry/Point2.h: In static member function ‘static gtsam::Vector gtsam::Point2::Logmap(const gtsam::Point2&)’:
/usr/local/include/gtsam/geometry/Point2.h:175:86: error: could not convert ‘Eigen::DenseBase<Derived>::operator<<(const Scalar&) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseBase<Derived>::Scalar = double]((& dp)->gtsam::Point2::x()).Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, 1> >((& dp)->gtsam::Point2::y())’ from ‘Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >’ to ‘gtsam::Vector {aka Eigen::Matrix<double, -1, 1>}’
   static inline Vector Logmap(const Point2& dp) { return (Vector(2) << dp.x(), dp.y()); }
                                                                                      ^
In file included from /usr/local/include/gtsam/base/SymmetricBlockMatrix.h:22:0,
                 from /usr/local/include/gtsam/linear/HessianFactor.h:21,
                 from /usr/local/include/gtsam/linear/GaussianFactorGraph.h:28,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizerParams.h:23,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizer.h:22,
                 from /usr/local/include/gtsam/nonlinear/LevenbergMarquardtOptimizer.h:21,
                 from /home/wangbx/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:41:
/usr/local/include/gtsam/base/SymmetricBlockMatrixBlockExpr.h: In instantiation of ‘class gtsam::SymmetricBlockMatrixBlockExpr<gtsam::SymmetricBlockMatrix>’:
/usr/local/include/gtsam/base/SymmetricBlockMatrix.h:118:62:   required from here
/usr/local/include/gtsam/base/SymmetricBlockMatrixBlockExpr.h:62:59: error: no type named ‘Index’ in ‘struct Eigen::internal::traits<gtsam::SymmetricBlockMatrixBlockExpr<gtsam::SymmetricBlockMatrix> >’
     typedef typename Eigen::internal::traits<This>::Index Index;
                                                           ^
In file included from /usr/local/include/gtsam/linear/HessianFactor.h:21:0,
                 from /usr/local/include/gtsam/linear/GaussianFactorGraph.h:28,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizerParams.h:23,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizer.h:22,
                 from /usr/local/include/gtsam/nonlinear/LevenbergMarquardtOptimizer.h:21,
                 from /home/wangbx/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:41:
/usr/local/include/gtsam/base/SymmetricBlockMatrix.h: In member function ‘gtsam::SymmetricBlockMatrix::Block gtsam::SymmetricBlockMatrix::operator()(gtsam::DenseIndex, gtsam::DenseIndex)’:
/usr/local/include/gtsam/base/SymmetricBlockMatrix.h:119:43: error: no matching function for call to ‘gtsam::SymmetricBlockMatrixBlockExpr<gtsam::SymmetricBlockMatrix>::SymmetricBlockMatrixBlockExpr(gtsam::SymmetricBlockMatrix&, gtsam::DenseIndex&, gtsam::DenseIndex&)’
       return Block(*this, i_block, j_block);
                                           ^
In file included from /usr/local/include/gtsam/base/SymmetricBlockMatrix.h:22:0,
                 from /usr/local/include/gtsam/linear/HessianFactor.h:21,
                 from /usr/local/include/gtsam/linear/GaussianFactorGraph.h:28,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizerParams.h:23,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizer.h:22,
                 from /usr/local/include/gtsam/nonlinear/LevenbergMarquardtOptimizer.h:21,
                 from /home/wangbx/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:41:
/usr/local/include/gtsam/base/SymmetricBlockMatrixBlockExpr.h:46:9: note: candidate: constexpr gtsam::SymmetricBlockMatrixBlockExpr<gtsam::SymmetricBlockMatrix>::SymmetricBlockMatrixBlockExpr(const gtsam::SymmetricBlockMatrixBlockExpr<gtsam::SymmetricBlockMatrix>&)
   class SymmetricBlockMatrixBlockExpr : public Eigen::EigenBase<SymmetricBlockMatrixBlockExpr<SymmetricBlockMatrixType> >
[... cut ...]
LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/build.make:62: recipe for target 'LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o' failed
make[2]: *** [LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/src/mapOptmization.cpp.o] Error 1
CMakeFiles/Makefile2:3321: recipe for target 'LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/all' failed
make[1]: *** [LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed

what's the problem and how to do?Thx

liuxiaonan123, 2018-10-09:
Hi, I have the same error, do you have known how to do ? Thx

ashnarayan13, 2018-10-13:
Hi, any update on this? Updating to boost 1.68 helped in building gtsam. But other packages fail which depend on boost 1.58. 
Is there a particular version/release of gtsam that was used to build lego loam?

liuxiaonan123, 2018-10-14:
Hi, I solved it by re-download gtsam,  maybe you can try it:
$ cd 
$ git clone https://bitbucket.org/gtborg/gtsam.git
$ cd gtsam
$ mkdir build
$ cd build
$ cmake ..
$ make check
$ make install

XuanboWang, 2018-10-15:
> Hi, any update on this? Updating to boost 1.68 helped in building gtsam. But other packages fail which depend on boost 1.58.
> Is there a particular version/release of gtsam that was used to build lego loam?

> Hi, any update on this? Updating to boost 1.68 helped in building gtsam. But other packages fail which depend on boost 1.58.
> Is there a particular version/release of gtsam that was used to build lego loam?

Hello, this problem has been solved. The solution is provided by liuxiaonan123. We only need to download the gtsam-4.0 source code and compile and install it to solve this problem.

XuanboWang, 2018-10-15:
> Hi, I solved it by re-download gtsam, maybe you can try it:
> $ cd
> $ git clone https://bitbucket.org/gtborg/gtsam.git
> $ cd gtsam
> $ mkdir build
> $ cd build
> $ cmake ..
> $ make check
> $ make install

Yes, this problem is solved by upgrading gtsam to version 4. Thank you very much.

closed 2018-10-15 by XuanboWang

grafoteka, 2019-03-27:
Hi all, 
I'm having the same error as @XuanboWang . 
I did what @liuxiaonan123 said. But I had to install it as admin.

Then, when I try to compile the ROS package I have the next error:

```
-- +++ processing catkin package: 'lego_loam'
-- ==> add_subdirectory(LeGO-LOAM/LeGO-LOAM)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:45 (message):
  Invalid arguments to find_dependency
Call Stack (most recent call first):
  /usr/local/lib/cmake/GTSAM/GTSAMConfig.cmake:17 (find_dependency)
  LeGO-LOAM/LeGO-LOAM/CMakeLists.txt:23 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/silth/WS/jaguar_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/silth/WS/jaguar_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

What can I do?

zuowanghanguang, 2019-03-28:
@grafoteka Hello.I have the same error as yours.Do  you have know how to solve it? Thanks.