RobustFieldAutonomyLab/LeGO-LOAM #33: How to modify parameters while using new lidar?

Zhehua-Hu, 2018-10-22:
I'm using a different 16 lines lidar, and I've set N_SCAN,ang_res_x y, but it doesn't work.

What does Horizon_SCAN mean? frequency * 360? Your paper shows Velodyne is 10Hz，360°

And What does  groundScanInd mean?
[image]
I set it to 16, and doesn't work.

Thanks for your work and help!

TixiaoShan, 2018-10-22:
@Zhehua-Hu , your lidar looks like sharing the same configurations as VLP-16 (-15~15), no changes need to be made.

Zhehua-Hu, 2018-10-23:
Thanks, I'll try it again.

zhehua@zhehua.info
 
From: Tixiao Shan
Date: 2018-10-23 07:02
To: RobustFieldAutonomyLab/LeGO-LOAM
CC: 哲华; Mention
Subject: Re: [RobustFieldAutonomyLab/LeGO-LOAM] How to modify parameters while using new lidar? (#33)
@Zhehua-Hu , your lidar looks like sharing the same configurations as VLP-16 (-15~15), no changes need to be made.

droneboxid, 2018-12-04:
hi @TixiaoShan please find link below of RS16 sample please can you try to run?

https://drive.google.com/file/d/1P_-ia5hm7eOOPLD9QIydfapllas2WpMg/view?usp=sharing

Thanks

theodosakis, 2018-12-25:
You can use the default settings of LeGO-LOAM.
The problem is because RS16 has NaN Points. To filter these you must run pcl filter.

```
<launch>
    <param name="/use_sim_time" value="true" />
   <node pkg="nodelet" type="nodelet" name="pcl_manager" args="manager" output="screen" />

  <!-- Run a passthrough filter to clean NaNs -->
  <node pkg="nodelet" type="nodelet" name="passthrough" args="load pcl/PassThrough pcl_manager" output="screen">
    <remap from="~input" to="/rslidar_points" />
    <remap from="/passthrough/output" to="/velodyne_points" />
    
    <rosparam>
      filter_field_name: z
      filter_limit_negative: True
    </rosparam>

   <rosparam>
      filter_field_name: x
      filter_limit_negative: True
    </rosparam>

   <rosparam>
      filter_field_name: y
      filter_limit_negative: True
    </rosparam>
  </node>
</launch>
```
Also, could you upload .bag file of RS16 Lidar with indoor building scanning?

rolandying, 2019-01-03:
> Thanks, I'll try it again.

Have you solve the kdtree error when using your own lidar? I have met the same problem. Looking forward for your reply.

droneboxid, 2019-01-03:
You can use the default settings of LeGO-LOAM.
The problem is because RS16 has NaN Points. To filter these you must run pcl filter.

```
<launch>
    <param name="/use_sim_time" value="true" />
   <node pkg="nodelet" type="nodelet" name="pcl_manager" args="manager" output="screen" />

  <!-- Run a passthrough filter to clean NaNs -->
  <node pkg="nodelet" type="nodelet" name="passthrough" args="load pcl/PassThrough pcl_manager" output="screen">
    <remap from="~input" to="/rslidar_points" />
    <remap from="/passthrough/output" to="/velodyne_points" />
    
    <rosparam>
      filter_field_name: z
      filter_limit_negative: True
    </rosparam>

   <rosparam>
      filter_field_name: x
      filter_limit_negative: True
    </rosparam>

   <rosparam>
      filter_field_name: y
      filter_limit_negative: True
    </rosparam>
  </node>
</launch>
```

rolandying, 2019-01-04:
> You can use the default settings of LeGO-LOAM.
> The problem is because RS16 has NaN Points. To filter these you must run pcl filter.
> 
> ```
> <launch>
>     <param name="/use_sim_time" value="true" />
>    <node pkg="nodelet" type="nodelet" name="pcl_manager" args="manager" output="screen" />
> 
>   
>   <node pkg="nodelet" type="nodelet" name="passthrough" args="load pcl/PassThrough pcl_manager" output="screen">
>     <remap from="~input" to="/rslidar_points" />
>     <remap from="/passthrough/output" to="/velodyne_points" />
>     
>     <rosparam>
>       filter_field_name: z
>       filter_limit_negative: True
>     </rosparam>
> 
>    <rosparam>
>       filter_field_name: x
>       filter_limit_negative: True
>     </rosparam>
> 
>    <rosparam>
>       filter_field_name: y
>       filter_limit_negative: True
>     </rosparam>
>   </node>
> </launch>
> ```

Thanks for your reply, it solved my problem! Good job, man!!
Thanks again! ^_^

NZK1993, 2019-01-14:
Hi！I do not know how to run  pcl filter ? Should I write a new ros node to remove the NaN points?

sujiongming, 2019-03-25:
@NZK1993  you should add the above lauch in the run.launch. such as bellow:
<launch>
    
    <include file="/home/ros/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/launch/above.launch" />
    
    
    <param name="/use_sim_time" value="true" />
    
    
    <node pkg="rviz" type="rviz" name="rviz" args="-d $(find lego_loam)/launch/test.rviz" />

    
    <node pkg="tf" type="static_transform_publisher" name="camera_init_to_map"  args="0 0 0 1.570795   0        1.570795 /map    /camera_init 10" />
    <node pkg="tf" type="static_transform_publisher" name="base_link_to_camera" args="0 0 0 -1.570795 -1.570795 0        /camera /base_link   10" />

        
    <node pkg="lego_loam" type="imageProjection"    name="imageProjection"    output="screen"/>
    <node pkg="lego_loam" type="featureAssociation" name="featureAssociation" output="screen"/>
    <node pkg="lego_loam" type="mapOptmization"     name="mapOptmization"     output="screen"/>
    <node pkg="lego_loam" type="transformFusion"    name="transformFusion"    output="screen"/>

</launch>

closed 2019-09-06 by stale[bot]

DeepDuke, 2020-04-05:
ang_res_x for RS lidar(RoboSense) is 0.18 ° if  rate is 10Hz, how did u set the parameters? Did u use the  optimized version of LeGO-LOAM? @Zhehua-Hu

w634307272, 2020-10-12:
> Thanks for your reply, it solved my problem! Good job, man!!
> Thanks again! ^_^

hello! I 've encountered same prolem as you did. I 'm wondering how can I run those code above, should I make a new XX.launch file in Lego_loam/launch/  and run XX.launch before I run run.launch, or do something else.