RobustFieldAutonomyLab/LeGO-LOAM #38: Adapt projecting pointcloud for different LIDAR sensor?

tuandle, 2018-11-05:
Hi @TixiaoShan,
Thanks for sharing this package. 
I have a question about the function `projectPointcloud`. In this [line](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/630a87bd446f859f616260f9729a4d0d1180e677/LeGO-LOAM/src/imageProjection.cpp#L205), you hardcoded two numbers for identifying the column index of a point:

```
if (horizonAngle <= -90)
        columnIdn = -int(horizonAngle / ang_res_x) - 450;
      else if (horizonAngle >= 0)
        columnIdn = -int(horizonAngle / ang_res_x) + 1350;
      else
        columnIdn = 1350 - int(horizonAngle / ang_res_x);
```
If I understand correct, those numbers `450,1350` are somewhat related to the `Velodyne VLP-16`. They will probably not fit for other LIDAR, whose horizontal resolution is not `0.2` as the `VLP-16`.
I wonder if you could help me understand the logic behind this these number so that I can adapt them to different LIDAR sensor?
I appreciate your help!

TixiaoShan, 2018-11-05:
@tuandle 
You are correct. If the resolution of the lidar is not 0.2, then you need to change the code. In this code, we wanted to let the point from the front of the lidar (12 o'clock) be at the center of the image. The point at 7 o'clock is at the left side of the image. The point at 5 o'clock is at the right side of the image. You don't have to obey this (front points projected to the center) as long as the points are correctly projected.

tuandle, 2018-11-06:
Thanks for your help!

tuandle, 2018-11-08:
Hi @TixiaoShan
I would like to ask for your help again. I adapted the function `projectPointCloud` to fit my lidar. However, I am not sure how to verify that I project it correctly.
Looking at your code, I have a question about the way you calculate the horizontal angle. 
[image]
In my drawing, if I understand correctly, you use `atan2(x,y)` [here](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/630a87bd446f859f616260f9729a4d0d1180e677/LeGO-LOAM/src/imageProjection.cpp#L202)  to calculate the horizontal angle for point p_i. According to my drawing, should we use `atan2(y,x)` to calculate the angle `gamma` as the horizontal angle for point p_i?
And how do we verify that the all the points are projected properly? I visualize the `rangeMat` image from your function and get this (I am using your `same_position.bag`):
[image]
So how do you know that you project the cloud correctly to `rangeMat`?
I appreciate your help!
Thanks again!

tuandle, 2018-11-12:
@TixiaoShan any help? I saw that you briefly mentioned the point cloud projecting quality while using a different LIDAR here #3 . 
Could you give me any hints how to access the projection quality?

TixiaoShan, 2018-11-17:
@tuandle Can you share a bag file? I will take a look into that.

tuandle, 2018-11-17:
@TixiaoShan Thanks for your help. Here is the [link](https://drive.google.com/file/d/1QUl3Wh1ka5ArJhzSYRzkkOTYVjSTult4/view?usp=sharing). The Lidar is Ouster OS1-64. 
The configuration for it is as folllows:
```
// Ouster 64
extern const int N_SCAN = 64;
extern const int Horizon_SCAN = 2000;
extern const float ang_res_x = 0.18;
extern const float ang_res_y = 0.52;
extern const float ang_bottom = 16.6 + 0.1;
extern const int groundScanInd = 20;
```

TixiaoShan, 2018-11-17:
@tuandle 
I am working on it. A quick question, is the terrain in the bag all flat, or it's a slope?

tuandle, 2018-11-17:
@tixiaoshan
I think it’s fairly flat because the bag is recorded with the lidar on top
of a car moving along a city road.

TixiaoShan, 2018-11-17:
@tuandle 
Do you have a bag that has IMU data in it? I know ouster lidar has built-in imu.

tuandle, 2018-11-17:
@TixiaoShan Yes, I obtained the same result. I will upload the bag with imu reading shortly. However, I doubt it can be useful because 1) the imu is a cheap one, so it doesn't have orientation readings; 2) I believe the imu coordinate is not aligned with the lidar.

tuandle, 2018-11-17:
@TixiaoShan Here is the new [link](https://drive.google.com/file/d/1TIXHClA3l__PnbGIxGludya4bBXMTsgX/view?usp=sharing) to the ouster bag with IMU readings.

rolandying, 2019-01-03:
> @TixiaoShan Thanks for your help. Here is the [link](https://drive.google.com/file/d/1QUl3Wh1ka5ArJhzSYRzkkOTYVjSTult4/view?usp=sharing). The Lidar is Ouster OS1-64.
> The configuration for it is as [...]

I have a 16-lidar with the same resolution, and I change the ang_res_x into 0.18, but I got a kdtree error:

`featureAssociation: /build/pcl-6_P28C/pcl-1.7.2/kdtree/include/pcl/kdtree/impl/kdtree_flann.hpp:136: int pcl::KdTreeFLANN<PointT, Dist>::nearestKSearch(const PointT&, int, std::vector<int>&, std::vector<float>&) const [with PointT = pcl::PointXYZI; Dist = flann::L2_Simple<float>]: Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to nearestKSearch!"' failed.
[featureAssociation-5] process has died [pid 1069, exit code -6, cmd /home/roland/catkin_ws/devel/lib/lego_loam/featureAssociation __name:=featureAssociation __log:=/home/roland/.ros/log/4400251c-0f2b-11e9-9666-303a64706e80/featureAssociation-5.log].
log file: /home/roland/.ros/log/4400251c-0f2b-11e9-9666-303a64706e80/featureAssociation-5*.log
`
How do you solve this problem, looking forward for your reply, thanks.

HappySamuel, 2019-03-26:
Hi @tuandle 

I tried to download your os1_imu.bag, however there's no TF inside the bag. Will result with errors.

closed 2019-09-06 by stale[bot]