RobustFieldAutonomyLab/LeGO-LOAM #43: LeGo-LOAM returns a wrong transformation when the car turns sharp at crossroads

Li-Shang, 2018-11-29:
I used LeGo-LOAM on *KITTI sequence 00*. when the car goes at crossroads, lego-loam returns wrong transformation as the image . [image]

I changed the iteration times from 25 to 50, but this doesn't work. I am confused about this.

mbudris, 2019-03-25:
Hi, I have the same behavior at sharp turns also. Tried to alter everything in the .h file, but without any considerable luck.

BrunoEduardoCSantos, 2019-04-05:
Hi,
Are you using IMU data? That may cause wrong transformations if IMU is not aligned with Velodyne axis.

mbudris, 2019-04-05:
It is level within a couple degrees.
I tried while being in still position, but a severe drift is observed. Without IMU there is much less drift, but still unable to tune to eliminate.

narutojxl, 2019-08-01:
Hi @brunoeducsantos i also meet this problem. After convert imu's orientation, acc and angular vel to imu' frame, which has the same origin with original imu frame but align with laser frame, the problem not solved. I test with kitti_2011_10_03_drive_0027_synced raw data.

closed 2019-11-05 by stale[bot]

xuwuzhou, 2019-12-25:
@narutojxl Hi,have you sovled your problem?I have met the same problem as you
[image]

J-Rojas, 2020-01-19:
I've seen the same problem outside of the scope of this project. If you map the path of the vehicle over time with the GPS/IMU pose data to the LiDAR point, there's a misalignment when the vehicle makes heading adjustments. After some time it realigns itself again. Maybe there is some unaccounted time lag in the GPS/IMU data?