RobustFieldAutonomyLab/LeGO-LOAM #44: odometry drift and the odometry is not the same plane?

Jaccica, 2018-12-08:
Hi. Thanks for your package.
I use the lego_loam with rslidar_points and changed the parameters in include file and the function in projectPointClolud, and there are some problems. First, the problem is the odometry always drift while our robot is static. Second,  the problem is the odometry is not the same plane. Can you expain why these problems happen? And can you give me some advices how to resolve these problems?
Thank you very much!

Jaccica, 2018-12-10:
@TixiaoShan Did you see my questions? Can you explain my doubts? Thanks.

Jaccica, 2018-12-21:
@NZK1993 hello. I don't solve the problem in essential. The encoder of velocity  is applied in this code, it will compute as long as the velocity is not zero, which is to solve the drift. Ａnd the resolution of second problem is to set the other coordinate values in height and the angle in roll and pitch zero. But the result is not good after changing.

zorosmith, 2019-04-04:
The accuracy of the lidar will affect the lidar odom.
I tested both robosense and velodyne-VLP16, I find that the the accuracy of point cloud from velodyne is heigher than robosense, especially for the object within 3 meter.
@Jaccica  Could you leave your email for me? I think we can have a disscusion.

lchop, 2019-04-24:
Hello,

Any progress on this issue ? I opened a new issue "Odom orientation issue", do you think it is the same issue ? Thank you !

closed 2019-09-06 by stale[bot]