RobustFieldAutonomyLab/LeGO-LOAM #49: IMU data error

louis960126, 2018-12-17:
My imu device is xsens mti-10 driver and driver http://wiki.ros.org/xsens_driver, 

I opened run.launch first and then open velodyne VLP16, it showed pointcloud on rviz.
But, when i use imu it will be stoped and show MSG to TF : Quaternion Not Properly Normalized.

thx for help!

wwtinwhu, 2018-12-22:
> My imu device is xsens mti-10 driver and driver http://wiki.ros.org/xsens_driver,
> 
> I opened run.launch first and then open velodyne VLP16, it showed pointcloud on rviz.
> But, when i use imu it will be stoped and show MSG to TF : Quaternion Not Properly Normalized.
> 
> thx for help!

hello,  I also met this question. Did you solve it now?

zorosmith, 2019-01-13:
Hello, could you show the /tf between velodyne and imu?
From the author's dataset, I can only know that the Z axis is pointing down.

louis960126, 2019-01-14:
Hi, I found that MTI-10 can't support orientation , so i use the [IMU-TOOLS](http://wiki.ros.org/imu_complementary_filter) to calculate the orientation.

wwtinwhu, 2019-01-14:
> Hi, I found that MTI-10 can't support orientation , so i use the [IMU-TOOLS](http://wiki.ros.org/imu_complementary_filter) to calculate the orientation.

thank you for your reply. I am also solved this problem. Did you have a better when use imu? I have a bad  result using imu. What about you? I don't know what is wrong. The imu frame did align with lidar

louis960126, 2019-01-14:
i have two result , one is small scenery ,the other is big scenery. in small scenery with or without IMU all looks good, but in big scenery it may drift more than 10m.  I am still trying kitti data set or some different scenery.

wwtinwhu, 2019-01-14:
> i have two result , one is small scenery ,the other is big scenery. in small scenery with or without IMU all looks good, but in big scenery it may drift more than 10m. I am still trying kitti data set or some different scenery.

We may have a same problem... May you leave a email so I can keep in touch with you?  The reason that when use imu some frames can't not match I think is the inital transform[6] is not correct. But why this inital transform from imu is not correct I can't figure out.

louis960126, 2019-01-14:
@wwtinwhu here is my E-mail : louis960126@gmail.com

BrunoEduardoCSantos, 2019-04-05:
@louis960126 you need to use a filter to compute orientation because Lidar IMU doesn't have orientation.

calvertdw, 2019-05-31:
> you need to use a filter to compute orientation because Lidar IMU doesn't have orientation

@BrunoEduardoCSantos Can you briefly describe what you mean? Can you link to some code or module that does the filtering?

I can't figure out how one "aligns" the IMU to the Lidar. Right now I have a LORD Microstrain 3DM-GX5-15 taped to the top of a VLP-16 on a cart and the result is slightly worse when I publish the IMU data.

HappySamuel, 2019-06-06:
> > Hi, I found that MTI-10 can't support orientation , so i use the [IMU-TOOLS](http://wiki.ros.org/imu_complementary_filter) to calculate the orientation.
> 
> Thank you for your reply. I also solved this problem. By the way, did you have a better result when use imu? I have a bad result using imu. What about you? I don't know what is wrong. The imu frame did align with lidar

HI @wwtinwhu 

Can you share your code on solving this problem? As i am using ouster lidar, which come along with built-in IMU, however it only provides 3 accelerations and 3 angular velocities. But lego-loam requires the orientation information from IMU.

Best Regards,
Samuel

closed 2019-09-06 by stale[bot]