RobustFieldAutonomyLab/LeGO-LOAM #50: Dense Final Cloud

athtest800, 2018-12-25:
Hi!
I would like to know if I can produce more dense final cloud.
Thank you.

droneboxid, 2018-12-26:
hi you can find at mapOptimization.cpp with edit parameter below;

        downSizeFilterCorner.setLeafSize(0.1, 0.1, 0.1);
        downSizeFilterSurf.setLeafSize(0.1, 0.1, 0.1);
        downSizeFilterOutlier.setLeafSize(0.1, 0.1, 0.1);

        downSizeFilterHistoryKeyFrames.setLeafSize(0.1, 0.1, 0.1);
        downSizeFilterSurroundingKeyPoses.setLeafSize(0.1, 0.1, 0.1);

        downSizeFilterGlobalMapKeyPoses.setLeafSize(0.1, 0.1, 0.1);
        downSizeFilterGlobalMapKeyFrames.setLeafSize(0.1, 0.1, 0.1);

and than rebuild catkin_make

zorosmith, 2019-04-04:
Hello, @droneboxid !
Do you know why the outlier point cloud is added to the final global map point cloud in _publishGlobalMap_ function in map Optmization.cpp. The code is shown in the following.
`for (int i = 0; i < globalMapKeyPosesDS->points.size(); ++i){
    int thisKeyInd = (int)globalMapKeyPosesDS->points[i].intensity;
    *globalMapKeyFrames += *transformPointCloud(cornerCloudKeyFrames[thisKeyInd],   &cloudKeyPoses6D->points[thisKeyInd]);
    *globalMapKeyFrames += *transformPointCloud(surfCloudKeyFrames[thisKeyInd],    &cloudKeyPoses6D->points[thisKeyInd]);
    *globalMapKeyFrames += *transformPointCloud(outlierCloudKeyFrames[thisKeyInd], &cloudKeyPoses6D->points[thisKeyInd]);
}`

chrisHuxi, 2019-04-14:
Hi, Did you find the answer? I have the same question. If you have solved it please tell me. : )
Thank you!

closed 2019-09-06 by stale[bot]