RobustFieldAutonomyLab/LeGO-LOAM #52: Run time error

SubMishMar, 2019-01-06:
I am able to install LeGO-LOAM without any error but when I roslaunch the run.launch file I get a memory leak or a run time error and here is the snapshot of the error.
[image]
I have done this on two computers and I get the same error everywhere. When I comment out the mapoptimization node in run.launch, I am able to run it without error but I need to run map optimization better results. Please help me figure this out.

SubMishMar, 2019-01-08:
I have tried this in a third computer. Same issue!

kimtj5521, 2019-04-11:
@SubMishMar 

open bashrc
$ gedit ~/.bashrc

add following line at the bottom of bashrc
export MALLOC_CHECK_=0

save and close bashrc
type this line on your terminal
$ source ~/.bashrc

after that, try again.

I solved same problem through this method.

---------------------------------
You can find easily what 'MALLOC_CHECK_=0' means.
do googling.
This is a trick to avoid 'double free of corruption' problem

facontidavide, 2019-08-02:
Do you still have this issue? Have you tried latest code on master? Have you tried to compile it with the AddressSanitizer on?

https://en.wikipedia.org/wiki/AddressSanitizer

wwtinwhu, 2019-08-05:
> @SubMishMar
>
> open bashrc
> $ gedit ~/.bashrc
>
> add following line at the bottom of bashrc
> export MALLOC_CHECK_=0
>
> save and close bashrc
> type this line on your terminal
> $ source ~/.bashrc
>
> after that, [...]

It works! Thank you very  much!  @kimtj5521  But I have a question that my computer didn't encounter this problem until today... It is fine ago, I don't know the reason why this double free of corruption happen. Do you know?

kimtj5521, 2019-08-05:
@wwtinwhu  I did not look at LeGO-LOAM's code in detail and could not find the exact cause yet. My colleagues also did not experience this problem. However, I encountered this problem and found the solution. 
If you find the cause, please let me know. Thanks.

facontidavide, 2019-08-05:
As I said, to precisely find what cause the error, Sanitizers can be used.
Once again: do you have a rosbag that can be used to reproduce the issue? Can you please share it?

kimtj5521, 2019-09-23:
@facontidavide Sorry for late reply. I tried again with the latest version of LeGO-LOAM and did not encounter this issue. (As of Sep. 23, 2019) 
An author's sample rosbag was used for the test.(2017-06-08-15-52-45_3.bag)
Also Sanitizer was not used.
In the latest code, the issue did not occur, so there was a little bug in the previous code or some miss in the process of using gtsam.
Anyway, there seems to be no problem with the LeGO-LOAM code of the current version.

TixiaoShan, 2019-11-19:
Glad the issue is solved.

closed 2019-11-19 by TixiaoShan