RobustFieldAutonomyLab/LeGO-LOAM #53: robosense laser mapping failed

improve100, 2019-01-11:
@TixiaoShan disturbed, the following two pictures were built with lego-loam, I tried to increase the imu data, but the result is the same as pure laser. (imu uses the data inside the gps module, the output frequency is 20hz) I don't know why when my car goes straight, it will go to the left, which will eventually lead to the failure of the entire map. The laser is robosense-16 and the laser output frequency is 10hz.
[image]
[image]

rosbag file:https://drive.google.com/file/d/1EwOSRwVtH1zGf58eZT6i2IZjZC43NY0c/view?usp=sharing

parameter：
Laser beam: 16 beams
Wavelength: 905nm
Laser class: class1
Accuracy: ±2cm(typical)
Range: 20cm to 150m(20% object reflectivity)
Data rate: 320，000pts/s
Field of view (vertical): +15.0° to -15.0° (30°)
Angular resolution(vertical): 2.0º
Field of view (horizontal): 360º
Angular resolution (horizontal/azimuth): 0.09º to 0.36º (5-20Hz)
RPM: 300rpm to 1200rpm(5Hz to 20Hz)
Input voltage: 9-32VDC
Power consumption: 9w(typical)
Sensor protection: IP67
Operation temperature: -10°C to +60°C
Dimension: H:82.7mm*φ:109mm
Weight: 0.84Kg(without cabling)
Data type: spatial coordinates/intensity

facontidavide, 2019-07-26:
Hi, 

since I am planning to use the Robosense LIDAR too, I have taken a look to your rosbag.

On my machine and using the latest version on master, I get better results than yours:

[image]

And this is the odometry 
[image]

Anyway, **the results are not fully deterministic.** Running again with the same executable and the same rosbag, I get these values:

[image]

facontidavide, 2019-08-12:
Your rosbag contains **/ns1/rslidar_points**  and  **/ns2/rslidar_points**.

Using **n1**, I get the same results that you get, using **n2** I get the better results that I have shown you.

closed 2019-10-18 by stale[bot]

HViktorTsoi, 2020-12-06:
Hi, we develop a tool that can convert Robosense point cloud to Velodyne format seamlessly, without modify the origin algorithm (you may only need to change the parameters like LiDAR beams, vertical resolution, etc.), we have tested it on LEGO-LOAM, LOAM and LIO-SAM.
You are welcomed to try this
[rs_to_velodyne](https://github.com/HViktorTsoi/rs_to_velodyne)

arjunskumar, 2022-02-18:
@facontidavide @improve100 could you share the bag file please?