RobustFieldAutonomyLab/LeGO-LOAM #56: Invalid arguments to find_dependency

zuowanghanguang, 2019-03-30:
Hello,when I run 'make -j1',I meet the error:
CMake Error at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:45 (message):
  Invalid arguments to find_dependency
Call Stack (most recent call first):
  /usr/local/lib/cmake/GTSAM/GTSAMConfig.cmake:17 (find_dependency)
  LeGO-LOAM/LeGO-LOAM/CMakeLists.txt:23 (find_package)
Does anyone known  the reason and how to solve it？Thank you!

TixiaoShan, 2019-03-31:
Have you installed GTSAM that is specified in readme successfully?

zuowanghanguang, 2019-04-01:
Yes.It is 4.0

mjjdick, 2019-04-09:
@zuowanghanguang I met the same error and solved by improved my cmake version to 3.10, maybe you can try this solution.

zuowanghanguang, 2019-04-09:
@mjjdick ok.I will try.Thank you!

WAT183, 2019-04-15:
I also met this error and I solved this by change "find_dependency" into "find_package" in file GTSAMConfig.cmake

JunGooLee, 2019-04-26:
I also meet same error. Could you let me know more details?
I couldn't find any ..... T.T

OneYouth, 2019-05-21:
> I also meet same error. Could you let me know more details?
> I couldn't find any ..... T.T
Both 'mjjdick' and 'WAT183' says are valid. The reason can be reference to 'https://github.com/pabloariasal/modern-cmake-sample/issues/5' , which says "find_dependency() does not support COMPONENTS argument before CMake 3.8.0".

closed 2019-09-06 by stale[bot]