RobustFieldAutonomyLab/LeGO-LOAM #58: How to do Localization only with LeGO-LOAM (Map provided)?

HappySamuel, 2019-04-06:
Hi @TixiaoShan 

I successfully generate map with your LeGO-LOAM in gazebo simulation, the map is nice. Here is the pointcloud map.
[image]

But if i want to use LeGO-LOAM to do the localization, is there anything need to be modified to allow localization in a known map? Can you guide me on this?

Best,
Samuel

TixiaoShan, 2019-04-06:
Hi @HappySamuel 
To localize with LOAM or LeGO-LOAM would need very heavy modifications to all the modules. For example, the feature maps are saved as a vector in mapRegistration.cpp. I would not suggest you do that. However, you may try segMap if you have an existing map.

HappySamuel, 2019-04-06:
Hi @TixiaoShan 

The segMap you mentioned, is it the one that did by ethz-asl? I will look into that and try it out.  Can I get back to you and consult you if i face any trouble on this segMap? As you have the experience with it.

Best,
Samuel

closed 2019-04-07 by TixiaoShan

Shelfcol, 2020-06-12:
Hi @TixiaoShan，have you tried relocalize the car using LeGO-LOAM with a  known lidar map? Could you give me some advice? thank you
looking forward to your reply.

tuencfs, 2021-02-17:
@Shelfcol @HappySamuel, I'm trying to do it also, did you find a solution for it?