RobustFieldAutonomyLab/LeGO-LOAM #59: Something wrong in linking Boost::Timer

linghusmile, 2019-04-07:
Hi all, I have meet the following error in catkin_make,I have seen other issue in this project for seeking something may seems like in same situtation but haven't found.  I doubt whether is the version of gtsam not match? But when i make install gtsam, nothing wrong happened. Any reply will be  helpful. thax a lot!
[ 73%] Linking CXX executable /home/chappie/catkin_ws/devel/lib/lego_loam/mapOptmization
/usr/bin/ld: 找不到 -lBoost::timer
collect2: error: ld returned 1 exit status
LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/build.make:791: recipe for target '/home/chappie/catkin_ws/devel/lib/lego_loam/mapOptmization' failed
make[2]: *** [/home/chappie/catkin_ws/devel/lib/lego_loam/mapOptmization] Error 1
CMakeFiles/Makefile2:1648: recipe for target 'LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/all' failed
make[1]: *** [LeGO-LOAM/LeGO-LOAM/CMakeFiles/mapOptmization.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed

linghusmile, 2019-04-07:
I have install sudo apt-get install libboost-all-dev.But seems doesn't works.

mbudris, 2019-04-07:
It is documented here:
http://wiki.ros.org/catkin/CMakeLists.txt#Boost

zorosmith, 2019-05-07:
I fixed this error.
Try to re-download the gtsam source code and reinstall it!

closed 2019-09-06 by stale[bot]