RobustFieldAutonomyLab/LeGO-LOAM #63: Parameters not used

BrunoEduardoCSantos, 2019-04-27:
Hi,
The following parameters are not being used:
* extern const int edgeFeatureNum = 2;
* extern const int surfFeatureNum = 4;
* extern const int sectionsTotal = 6;
Could you explain me if they have some other purpose?
Thanks,
Bruno

zorosmith, 2019-05-10:
Hello @BrunoEduardoCSantos !
I found that these 3 parameter are used in the extractFeatures function in featureAssociation.cpp. Relative codes are showed in the following.
`for (int j = 0; j < 6; j++)`    for 6 sub-images;
`if (largestPickedNum <= 2)`    for 2 edge feature points in one sub-image;
`if (smallestPickedNum >= 4)`    for 4 surf feature points in one sub-image;
Hope it helps you.

BrunoEduardoCSantos, 2019-05-10:
Thanks @zorosmith . It clears out my mind!

closed 2019-05-10 by BrunoEduardoCSantos