RobustFieldAutonomyLab/LeGO-LOAM #68: roslaunch error

stooooooone, 2019-05-29:
roslaunch lego_loam run.launch, then ...

01:~$ roslaunch lego_loam run.launch
... logging to /home/dfg/.ros/log/02ab6d3e-7e3d-11e9-ad23-000babd9c1eb/roslaunch-dfg-01-19532.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is &lt;1GB.
 
started roslaunch server http://dfg-01:42585/
 
SUMMARY
========
 
PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True
 
NODES
  /
    base_link_to_camera (tf/static_transform_publisher)
    camera_init_to_map (tf/static_transform_publisher)
    featureAssociation (lego_loam/featureAssociation)
    imageProjection (lego_loam/imageProjection)
    mapOptmization (lego_loam/mapOptmization)
    rviz (rviz/rviz)
    transformFusion (lego_loam/transformFusion)
 
auto-starting new master
process[master]: started with pid [19542]
ROS_MASTER_URI=http://localhost:11311
 
setting /run_id to 02ab6d3e-7e3d-11e9-ad23-000babd9c1eb
process[rosout-1]: started with pid [19555]
started core service [/rosout]
process[rviz-2]: started with pid [19562]
process[camera_init_to_map-3]: started with pid [19573]
process[base_link_to_camera-4]: started with pid [19581]
process[imageProjection-5]: started with pid [19589]
process[featureAssociation-6]: started with pid [19605]
process[mapOptmization-7]: started with pid [19619]
process[transformFusion-8]: started with pid [19634]
[ INFO] [1558713612.116897928]: ----&gt; Image Projection Started.
[ INFO] [1558713612.121958607]: ----&gt; Feature Association Started.
[ INFO] [1558713612.139426433]: ----&gt; Transform Fusion Started.
[mapOptmization-7] process has died [pid 19619, exit code -11, cmd /home/dfg/catkin_ws/devel/lib/lego_loam/mapOptmization __name:=mapOptmization __log:=/home/dfg/.ros/log/02ab6d3e-7e3d-11e9-ad23-000babd9c1eb/mapOptmization-7.log].
log file: /home/dfg/.ros/log/02ab6d3e-7e3d-11e9-ad23-000babd9c1eb/mapOptmization-7*.log

stooooooone, 2019-05-29:
ask for help, thanks anyway

TixiaoShan, 2019-05-30:
@stooooooone 
That might be a GTSAM problem. Please follow the installation instructions provided in the README and install again.

https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/master/README.md

stooooooone, 2019-05-31:
> @stooooooone
> That might be a GTSAM problem. Please follow the installation instructions provided in the README and install again. Let me know if it works.
> 
> https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/master/README.md

i used the wrong version of the gtsam
now, it works ! 
thank u very much !

closed 2019-05-31 by TixiaoShan

nyangshawbin, 2020-05-25:
Hi @TixiaoShan,

I experienced the same issue as @stooooooone. I had followed the README and tried reinstalling once but the same error message still appeared as follows:

[image]

My version is gtsam-4.0.0-alpha2 which seems to be identicial to what was mentioned in the README.

arm3nyn, 2020-07-17:
> Hi @TixiaoShan,
> 
> I experienced the same issue as @stooooooone. I had followed the README and tried reinstalling once but the same error message still appeared as follows:
> 
> [image]
> 
> My version is gtsam-4.0.0-alpha2 which seems to be identicial to what was mentioned in the README.

+1 same here

nyangshawbin, 2020-07-18:
could try re-installing from scratch. my problem was resolved after i redid