RobustFieldAutonomyLab/LeGO-LOAM #72: mapOptimization shutdown unexpectedly with errors and warnings

HappySamuel, 2019-05-31:
Hi

I tried with Ouster 1-16 Lidar, I record a rosbag and used it. I comment the vlp-16 configuration and uncomment the os1-16 configuration in utility.h

However when i run both the run.launch and bagfile. I got the following errors.

[image]

And this error eventually lead to the shutdown of mapOptimization node.

[image]

I have no idea where the errors and warnings come from.
Can you please help me out?

Here i attached the link to the bag file:
https://drive.google.com/open?id=1ysiFH0WhU6JqU-xnwHeLL13Tipy9G0JH

Best Regards,
Samuel

TixiaoShan, 2019-06-02:
Hi @HappySamuel 

Are you able to run it at all? I have no problem running your bag.
[image]
[image]

HappySamuel, 2019-06-02:
Hi @TixiaoShan 

I tried again, but still the same error. Can you share your configuration for running it? Maybe i missed something? Like launch file, utility.h, or command run on terminal?

Best,
Samuel

TixiaoShan, 2019-06-02:
@HappySamuel 
I just made a few changes. Take a look at the commits made today.
My command:
`roslaunch lego_loam run.launch`
`rosbag play 2019-05-27-18-38-00.bag --clock --topic /os1_cloud_node/points`

TixiaoShan, 2019-06-02:
@HappySamuel 
Ah, I think the imu data from Ouster lidar is different from the data from some other mainstream IMUs. I had that problem before. I will take a look at the imu data in detail once I have more time. You can just publish point cloud data for now to use LeGO-LOAM.

HappySamuel, 2019-06-03:
Hi @TixiaoShan 

I did as what you said, but how come the map i generated is different from yours? Here i attached the map below:
[image]

Best,
Samuel

TixiaoShan, 2019-06-04:
@HappySamuel 
That looks weird. The point cloud map looks ok. The trajectory is completely wrong.

HappySamuel, 2019-06-04:
Hi @TixiaoShan 

Could it possibly be the mapping parameters that give different result? Maybe mine and yours are different? But i am using your latest update package, however no matter how many times i tried, i always got the same result.

Best,
Samuel

HappySamuel, 2019-06-04:
Hi @TixiaoShan 

I manage to get the same result as yours. It's due to i comment out the VLP-16 parameters in utility.h, but i forgot to uncomment the OS1-16 parameters. After i uncomment them, the result is correct now.

[image]

Appreciate for your help and patient that you gave. Thank you very much.
 
Best, 
Samuel

closed 2019-06-04 by HappySamuel