RobustFieldAutonomyLab/LeGO-LOAM #74: Adapting LeGO-LOAM for different LiDARS

Ashik19, 2019-06-01:
Hi,
I am trying to use the LeGO-LOAM tool for mapping purpose using the Ouster(OS1_16) LiDAR. Can anyone please tell me what I should change apart from the Utility.h, imageprojection.cpp file to adjust the configuration?

Thanks
Ashik

TixiaoShan, 2019-06-01:
Hi @Ashik19 
Please refer to this question to use OS-1-16.

https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/70

Ashik19, 2019-06-01:
Thank you very much for your kind reply @TixiaoShan. But unfortunately, when I am running the Lego loam it seems like when I am using the odom_ laser frame it moves by itself in a circular manner even though I do not move my LiDAR unit. I guess apart from the utility.h and imageprojection.cpp there are some other settings that need to be changed to work with ouster LiDAR. Can you please help me with this issue.

Ashik19, 2019-06-01:
And one more question whenever I am trying to record a rosbag with point cloud2 topic(/os1_cloud_node/points) using ouster os1-16 LiDAR. Irrespective  of my recording duration whenever I convert my bag into pcd I always get the same number of data points. Can you please tell me where I am making mistakes? Also it seems like recording the pointcloud2 gets a very big bag file size within the short time. My final goal is to make a map using the LiDAR. It would be very kind of you if you could help me with this issue.

TixiaoShan, 2019-06-02:
@Ashik19 
I just ran a bag file that was posted in `https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/72`

Maybe you can take a look at that bag file. I guess something is wrong with your data.

closed 2019-06-02 by Ashik19