RobustFieldAutonomyLab/LeGO-LOAM #79: Invalid argument "/camera_init" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like:

HappySamuel, 2019-06-04:
Hi @TixiaoShan 

I tried to use pointcloud_to_pcd function from pcl_ros package to save the map. But I found that there might be some issue for using "/" on declaration of the frame id, especially in tf2. The error is attached as below:

[image]

However, if the "/" is removed from declaration of the frame id, won't have problem like this anymore.

Best Regards,
Samuel

TixiaoShan, 2019-06-04:
@HappySamuel 
Thanks for point this out!

zorosmith, 2019-06-05:
When I save the map, I did not meet this error.

HappySamuel, 2019-06-12:
Hi @zorosmith 

With the new map saving function that TixiaoShan added in, this error is gone.

Best,
Samuel

closed 2019-06-12 by HappySamuel