RobustFieldAutonomyLab/LeGO-LOAM #80: Error in Rosbag play

Ashik19, 2019-06-04:
Dear Altruist people,

When I am trying to play the pointcloud2 topic in the legloam of a pre-recorded bag file with "map" as the fixed frame I am getting an error

"Message_Removed because its too old(frame=/os1_lidar) stamp=[587.990

and a warning under the TF for aft_map

[image]

Can anyone please tell me where I am making mistake ?

I have changed my launch file, utility.h, and imageprojection.cpp file to adjust the change for using the LeGO-LOAM for Ouster (OS_16). The herewith attached picture is for my launch file which modified from other users recommendation.
[image]

Also, can anyone tell me is there anyway to get a single pcd file containing all the pointcloud of a bag file ? Or how can I get the laz file from a bag file ?

Thank you very much for your valuable time.

Thanks

Dinatorehanova, 2019-06-07:
Hi @TixiaoShan,

I want to share something about my tf tree to clarify my problem. its like when I initially start the lego-loam my tf tree is like as following 

[image]

But once I run the rqt_tf_tree then it gets changed along with a warning like

[image]

[image]

I have not even opened my pointcloud2 topic. I guess I am missing something important.I am using the latest package of lego_loam and made all the required change in the following files (utility.h, imageprojection.cpp, run.launch) to adjust for ouster(os_16).

Thank you very much for your valuable time.

Thanks 

Ashik

Ashik19, 2019-06-07:
Hi @TixiaoShan,

I want to share something about my tf tree to clarify my problem. its like when I initially start the lego-loam my tf tree is like as following 

[image]

But once I run the rqt_tf_tree then it gets changed along with a warning like

[image]

[image]

I have not even opened my pointcloud2 topic. I guess I am missing something important.I am using the latest package of lego_loam and made all the required change in the following files (utility.h, imageprojection.cpp, run.launch) to adjust for ouster(os_16).

Thank you very much for your valuable time.

Thanks 

Ashik

Ashik19, 2019-06-08:
I solved this one.The pitfall was I uncomment the cloudheader.stamp=ros::Time::now(); in the image projection.cpp file.By doing uncomment it turns back the timestamp causing it too old to publish the pointcloud topic

closed 2019-06-08 by Ashik19