RobustFieldAutonomyLab/LeGO-LOAM #91: Mapping Issue with bag file

Ashik19, 2019-06-21:
Hi @TixiaoShan ,

Can you please have a look at the herewith attached bag file ? I am not getting any trajectory data and the map it is generating is very odd compared to the ground.I collected this ata using ouster lidar mounted on a car.I have made change in the utility.h file in order to adjust the relative change but still do not know why I am getting this error.

https://l.facebook.com/l.php?u=https%3A%2F%2Fdrive.google.com%2Ffile%2Fd%2F1BvZiqFwOI9FRk23IK5fU5Oxav6dzWe2y%2Fview%3Fusp%3Dsharing%26fbclid%3DIwAR00pFBehju7tip2cu536zzG-woLD7Q3FYmRdfMG0Ohfb-3FgjU3XKEDLeI&h=AT3Ef6f0J9lUOFcp7PsoXu-mXKNlJhwfRJ_GORz_ENLc_4NkO8TNdiGiNZIQtpHATgzm2tMol3qvuBj6ZVb8J0NfzvU9YeHDO3DPirIKK4NAJOuXDxNx0cjOk84p-w

[image]

Thank you very much for your valuable time. I will wait for your  valuable feedback.

TixiaoShan, 2019-06-22:
Hi @Ashik19 
It seems your point cloud data is wrong. The white line in the image represents the horizon. It seems that all your data is offset downwards.

[image]

Ashik19, 2019-06-23:
Hi, @TixiaoShan ,

Thank you very much for your kind reply.I have also faced this same issue with some of my other bag files like this.I am using an ouster Lidar(OS-16) mounted on a car. Can you recommend something about how to avoid this issue? 
Thanks

TixiaoShan, 2019-06-23:
Hi @Ashik19 
I can't comment on that. I never used that lidar. Probably it's best to ask the tech support.

Ashik19, 2019-06-23:
@tixiaoshan, thank you very much for being patient with me. Can you please tell me how did you get the white line indicating the horizon of the point cloud ?

Ashik19, 2019-06-23:
Hi @ tixiaoshan,

Could it be possible that I have to use a horizontal resolution of 1024 always while using Lego loam in order to resolve this problem. I guess the code in the utility.h looks for a horizontal scan resolution of 1024 while making maps. There are different resolution and scan speed available in ouster like “512x10”, “512x20”,”1024x10”, and “1024x20”. I guess if I wang to use different resolution each time I will have to change the parameter in the utility.h and do the catkin_make? 

Thank you for your valuable time

TixiaoShan, 2019-06-24:
Yeah, the params in utility.h are set for different sensors with different resolutions. You need to re-compile for each resolution.

TixiaoShan, 2019-06-24:
> @TixiaoShan, thank you very much for being patient with me. Can you please tell me how did you get the white line indicating the horizon of the point cloud ?

It's just the Grid in Rviz.

Ashik19, 2019-06-24:
Hi @tixiaoshan,

Thank you very Much for your kind reply. I figured the location of the lidar matters while mounting it on car. Also may I know whether I will have to change the Other parameter like angular resolution  x,y  in the utility.h file since I am using a car to work with LeGO- Loam.

TixiaoShan, 2019-06-24:
This user's bag has the same issue like yours (point cloud not symmetric relative to the horizon).

https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/94

closed 2019-06-30 by Ashik19