RobustFieldAutonomyLab/LeGO-LOAM #92: Map optimization process died when launching lego loam

Dinatorehanova, 2019-06-21:
Hi @TixiaoShan 

Can you please have a look at the herewith attached picture ?
[image]

How can I resolve this issue ?

Thanks

W-Liur, 2019-06-22:
Have you solved it? I have the same problem.

Ashik19, 2019-06-22:
@W-Liur , No, I do not understand why it's coming up. I have tried to clone and do the catkin_make again .But still no success. Please let me know if you come up with any solution.

Dinatorehanova, 2019-06-22:
@W-Liur , No I have not solved it yet. I have tried to clone and do the catkin_make -j1 again but still with no success.Please, let me know if you have any update on this.

Thanks

W-Liur, 2019-06-22:
Nor did I succeed.I'm trying the previous version, but it's a bit slow to download.

W-Liur, 2019-06-22:
I have fixed it.You can reinstall gtsam.Maybe you need this version.https://pan.baidu.com/s/1wVwtAT-Q3uD2WZN9xoiMBg      KEY:i64o

TixiaoShan, 2019-06-22:
Looks like it's a GTSAM problem. Have you tried the version that is used in README?

Dinatorehanova, 2019-06-22:
Hi, @TixiaoShan, Thank you very much for your kind reply. I tried to install the mentioned GTSAM and it works.But still there is 

"0x2bf88d0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x2557460) ): Attempt to set a screen on a child window."

[image]

Can you please tell me how to address this ? Also I would like to know whether there is any speed issue while using the LeGO-LOAM as it seems id does not provide any map or clear trajectory while using this tool for a high speed car with LiDAR mounted on it.

Thanks

TixiaoShan, 2019-06-22:
I don't know this new issue. It doesn't look like a LeGO-LOAM issue.
If the car speed is too high, there might be not enough overlaps between two scans, then LOAM will fail.

Dinatorehanova, 2019-06-23:
Thank you @TixiaoShan .

closed 2019-06-23 by Dinatorehanova