RobustFieldAutonomyLab/LeGO-LOAM #94: No trajectory path while recording bag file in (sim time False mode)

Dinatorehanova, 2019-06-23:
Hi, @TixiaoShan  I was trying to make an indoor map with sim time false but when I was moving with my lidar unit inside the room it does not provide me any trajectory path.I also tried to again play the bag file in simulation time(true) but still it does not provide any trajectory.Can you please spare some of your valuable time to look into the herewith attached bag file 

https://drive.google.com/file/d/18mdPdLWP-nrtKwTwl6-D9ZCPOx2OD5lv/view?usp=sharing

[image]

Thank you very much for your valuable time.

TixiaoShan, 2019-06-24:
It seems that you have the same problem like this issue.

https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/91

Dinatorehanova, 2019-06-24:
@Tixiaoshan,thanks for your valuable feedback. I am encountering an error like even when my vehicle is static it seems the legoloam (sim time false) is moving my base_link and creating map with very strange trajectory and generating maps which are very weird compared to my surround. Can you please tell me what can cause this issue ?

Thank you

TixiaoShan, 2019-06-24:
So I assume you have solved https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/91. That's weird it's still not working.

Dinatorehanova, 2019-06-24:
Thank you @TixiaoShan . It was my mistake I tried different resolution without configuring the utility.h file

closed 2019-06-24 by Dinatorehanova

ghz009, 2021-05-06:
> Thank you @TixiaoShan . It was my mistake I tried different resolution without configuring the utility.h file

Hi @Dinatorehanova ，I met the same problem. Can you tell me how you solved the problem，please.

[image]