RobustFieldAutonomyLab/LeGO-LOAM #96: Map optimization wants topic /odometry/filtered to have datatype/md5sum[sensor_msgs/IMU/] dropping connection

Ashik19, 2019-06-24:
[image]

@TixiaoShan , I am trying to use the robot localization package and use the /odometry/filtered topic generated from the EKF(extended kalman filter node) as the imu topic in the utility.h file.But somehow I am coming up with the given error.Can you please tell me how to address this problem ?

Thank you very much for your valuable time.

Best
Ashik

HappySamuel, 2019-06-25:
Hi @Ashik19 

From what i know, "odometry/filtered" is originally from Robot Localization Package, and its message type is nav_msgs::Odometry. And what featureAssociation and mapOptimization are requesting sensor_msgs::Imu. So you need to give it IMU topic, instead of odometry topic.

Best,
Samuel

chenweng2009, 2019-06-25:
Hi @HappySamuel ,

Thank you very much for your kind reply. I have an INS unit which publishes /ins and /imu topic. 

[image]

When I am putting the /ins topic in the utility.h getting the following error

[image]
And when I am putting the /imu topic in the utility.h I am getting this error

[image]

@TixiaoShan, can you please suggest me how I can address this problem ?

Thank you very much for your valuable time.

Ashik19, 2019-06-25:
Hi @HappySamuel ,

Thank you very much for your kind reply. I have an INS unit which publishes /ins and /imu topic. 
The /ins is (navigation odometry) while the /imu is the imu.
[image]

The data provided by this 2 topics are following
[image]

When I am putting the /imu topic in the utility.h I am getting the following error
[image]

@TixiaoShan, can you please suggest me how I can address this problem ?

Thank you very much for your valuable time.

closed 2019-06-30 by Ashik19

HappySamuel, 2019-07-01:
Hi @Ashik19 

Sorry for the late reply. The imu data stream that LeGO-LOAM package required is the orientation section. However, your /imu data stream is having the orientation all zeros. That's the reason causes the error message.

Best,
Samuel