RobustFieldAutonomyLab/LeGO-LOAM #104: frames meaning

narutojxl, 2019-07-12:
Hello @TixiaoShan 

when launching lego_loam and play [jackal_dataset_20170608 bags](https://github.com/RobustFieldAutonomyLab/jackal_dataset_20170608), i got the frame tree as follows.
[image] In the picture,  the meaning of virtual_velodyne frame is (Z-front, X-left, Y-up), which different from velodyne_frame(X-front, Y-left, Z-up). 

Q1. Does base_link in tf tree meaning velodyne_frame?  camera is virtual_velodyne? 

Q2. What's the meaning of  lego_loam backend calculated frame aft_mapped?   I see transformAftMapped's orientation is transformed before published, but when saving finalCloud.pcd map, use cloudKeyPoses6D(cloudKeyPoses6D == transformAftMapped) to transform points to camera_init frame, not use aft_mapped or camera frame.

>      
       void publishTF(){
        geometry_msgs::Quaternion geoQuat = tf::createQuaternionMsgFromRollPitchYaw
                                  (transformAftMapped[2], -transformAftMapped[0], -transformAftMapped[1]);
        odomAftMapped.header.stamp = ros::Time().fromSec(timeLaserOdometry);
        odomAftMapped.pose.pose.orientation.x = -geoQuat.y;
        odomAftMapped.pose.pose.orientation.y = -geoQuat.z;
        odomAftMapped.pose.pose.orientation.z = geoQuat.x;
        odomAftMapped.pose.pose.orientation.w = geoQuat.w;
        odomAftMapped.pose.pose.position.x = transformAftMapped[3];
        odomAftMapped.pose.pose.position.y = transformAftMapped[4];
        odomAftMapped.pose.pose.position.z = transformAftMapped[5];
        odomAftMapped.twist.twist.angular.x = transformBefMapped[0];
        odomAftMapped.twist.twist.angular.y = transformBefMapped[1];
        odomAftMapped.twist.twist.angular.z = transformBefMapped[2];
        odomAftMapped.twist.twist.linear.x = transformBefMapped[3];
        odomAftMapped.twist.twist.linear.y = transformBefMapped[4];
        odomAftMapped.twist.twist.linear.z = transformBefMapped[5];
        pubOdomAftMapped.publish(odomAftMapped);
        aftMappedTrans.stamp_ = ros::Time().fromSec(timeLaserOdometry);
        aftMappedTrans.setRotation(tf::Quaternion(-geoQuat.y, -geoQuat.z, geoQuat.x, geoQuat.w));
        aftMappedTrans.setOrigin(tf::Vector3(transformAftMapped[3], transformAftMapped[4], transformAftMapped[5]));
        tfBroadcaster.sendTransform(aftMappedTrans);
    }

>     
        thisPose3D.x = latestEstimate.translation().y();
        thisPose3D.y = latestEstimate.translation().z();
        thisPose3D.z = latestEstimate.translation().x();
        thisPose3D.intensity = cloudKeyPoses3D->points.size(); // this can be used as index
        cloudKeyPoses3D->push_back(thisPose3D);
        thisPose6D.x = thisPose3D.x;
        thisPose6D.y = thisPose3D.y;
        thisPose6D.z = thisPose3D.z;
        thisPose6D.intensity = thisPose3D.intensity; // this can be used as index
        thisPose6D.roll = latestEstimate.rotation().pitch();
        thisPose6D.pitch = latestEstimate.rotation().yaw();
        thisPose6D.yaw = latestEstimate.rotation().roll(); // in camera frame
        thisPose6D.time = timeLaserOdometry;
        cloudKeyPoses6D->push_back(thisPose6D);
        /**
         * save updated transform
         */
        if (cloudKeyPoses3D->points.size() > 1) {
            transformAftMapped[0] = latestEstimate.rotation().pitch();
            transformAftMapped[1] = latestEstimate.rotation().yaw();
            transformAftMapped[2] = latestEstimate.rotation().roll();
            transformAftMapped[3] = latestEstimate.translation().y();
            transformAftMapped[4] = latestEstimate.translation().z();
            transformAftMapped[5] = latestEstimate.translation().x();

            for (int i = 0; i < 6; ++i) {
                transformLast[i] = transformAftMapped[i];
                transformTobeMapped[i] = transformAftMapped[i];
            }
        }

narutojxl, 2019-07-15:
when i try to publish delta_TF between  transformAftMapped and aftMappedTrans, the delta_transform is not a const value, does anybody know why or something  wrong?  Here is the test code added in    [mapOptmization.cpp::publishTF()](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/master/LeGO-LOAM/src/mapOptmization.cpp#L652)

>          
        // jxl  transformAftMapped TF
        //transformAftMapped[0 1 2 3 4 5] 分别一一对应[x_pitch y_yaw z_roll x y z]
        tf::StampedTransform keyFramesTrans;
        keyFramesTrans.frame_id_ = "/camera_init";
        keyFramesTrans.child_frame_id_ = "/keyFrames";
        keyFramesTrans.stamp_ = ros::Time().fromSec(timeLaserOdometry);
        geoQuat = tf::createQuaternionMsgFromRollPitchYaw(transformAftMapped[0], transformAftMapped[1], transformAftMapped[2]);
        keyFramesTrans.setRotation(tf::Quaternion(geoQuat.x, geoQuat.y, geoQuat.z, geoQuat.w));
        keyFramesTrans.setOrigin(tf::Vector3(transformAftMapped[3], transformAftMapped[4], transformAftMapped[5]));
        tfBroadcaster.sendTransform(keyFramesTrans);

        //一个姿态可以由不同的rpy表示
        //TF: keyFrames---> aftMapped
        tf::Transform delta_tf;
        delta_tf = keyFramesTrans.inverseTimes(aftMappedTrans);
        tf::Vector3 delta_trans;
        tf::Quaternion delta_orientation;
        double delta_roll, delta_pitch, delta_yaw;
        delta_trans = delta_tf.getOrigin();
        delta_orientation = delta_tf.getRotation();
        tf::Matrix3x3(delta_orientation).getRPY(delta_roll, delta_pitch, delta_yaw);
        delta_roll *= RAD2DEGREE;
        delta_pitch *= RAD2DEGREE;
        delta_yaw *= RAD2DEGREE;

        geometry_msgs::PointStamped diff_msg;
        diff_msg.header.frame_id = "camera_init";
        diff_msg.header.stamp = ros::Time().fromSec(timeLaserOdometry);
        diff_msg.point.x = delta_roll;
        diff_msg.point.y = delta_pitch;
        diff_msg.point.z = delta_yaw;
        pub_diff_keyframes_to_aft_mapped.publish(diff_msg);
        //std::printf("diff_keyframs_aftmapped: %f, %f, %f \n", delta_roll, delta_pitch, delta_yaw);
[image]

closed 2019-08-02 by narutojxl

tejalbarnwal, 2022-03-29:
@narutojxl did you get the answers to the questions you had posted?