RobustFieldAutonomyLab/LeGO-LOAM #110: Terrible performance on the dataset you shared using LOAM

LiShuaixin, 2019-07-31:
Hi, @TixiaoShan . I want to test LOAM with the dataset you shared on your Google Cloud to compare its performance with LeGO-LOAM as you did in your paper.

I found that LOAM works terribly with your dataset as long as it makes a large turn. But I also found that LOAM generated acceptable results in your paper. Would you please give me a hint to test LOAM with your dataset?  I just run LOAM and bag file in two different ternimals using roslaunch and rosbag tool.

Scrumantic, 2019-08-05:
I've also done some benchmarking and used LOAM for a while before discovering LeGO-LOAM. The only way I managed to process large and complex bags was to decrease the rate which rosbag plays. I don't know if that helps but if you add -r 0.10 (10% speed) to your rosbag play command your results might improve a lot. (sorry if this isn't helpful)

LiShuaixin, 2019-08-06:
@Scrumantic Hi, I tried another version of LOAM named A-LOAM, which uses some open source library like eigen and ceres to replace cv::Mat to improve system operation efficiency. I seems much better.

closed 2019-10-12 by stale[bot]