RobustFieldAutonomyLab/LeGO-LOAM #112: About the rate factor of rosbag play

HViktorTsoi, 2019-08-02:
Is there a implied condition that the rate(factor) of `rosbag play` must be 1? 

When I tried to speed up the running time by using a rosbag play rate higer than 1 (e.g. -r=20), there is serious bias in the results, as shown in following figures.

[image]
_fig.1. rosbag play -r=1 (normal)_

[image]
_fig.2. rosbag play -r=20  (wrong)_

So is there a implied require that the rate of `rosbag play` should be 1?  Or is there a way to increase the play rate without hurting the performance? Thanks a lot.

facontidavide, 2019-08-02:
Short answer : yes, use rate 1

Long answer: not necessarily 1, but not 20 for sure.

The software is implemented with 3 individual nodes. You need to allow enough time to each node to perform its computation or some measurements will be "skipped".

This is strongly related to issue #107

Scrumantic, 2019-08-05:
I've processed forestry acquired bags at 10x rate without problems. Although for my solution I limit the VLP16 range to 50 meters.

facontidavide, 2019-08-13:
Hi. 

I have solved this issue... doing a heavy refactoring of the code. Please refer to the comment in #107

closed 2019-10-15 by HViktorTsoi