RobustFieldAutonomyLab/LeGO-LOAM #115: project the point cloud map to google earth

weisongwen, 2019-08-08:
Hi @TixiaoShan , Thanks for sharing your work. How do you project the poitn cloud map into the google earth which can be seen in your published paper?

[image]

many thanks,
Welson,

mbudris, 2019-08-08:
https://github.com/gareth-cross/rviz_satellite
can try this one

Ashik19, 2019-08-12:
Hi @weisongwen ,

Can you share which platform you used to georeference the point cloud so that it overlaps with the real world place ?

Thanks

weisongwen, 2019-08-13:
@mbudris  Many Thanks,
@Ashik19 I expect to use google earth.

Ashik19, 2019-08-14:
Hi @weisongwen ,

Can you share your work on georeferencing point cloud ? I am stuck with how I can transfer the LiDAR local coordinates into my gps coordinates(WGS-1984) so that it overlaps with the real world place.

Thanks

weisongwen, 2019-08-15:
@Ashik19  you can refer to my code in [here](https://github.com/weisongwen/GNSS-INS/blob/master/amsi/include/amsi/gnss_tools.hpp).

simon-rob, 2019-09-13:
@weisongwen,

I am also stuck on transforming local coords to gps coords. I tried to follow the link you posted but it is not valid.

Many thanks,

Simon

weisongwen, 2019-09-14:
@simon-rob , please try [this](https://github.com/weisongwen/amsi/blob/master/amsi/include/amsi/gnss_tools.hpp). 
I suggest you  firstly transfer the odometry from LiDAR odometry into the [ENU coordiate](https://en.wikipedia.org/wiki/Local_tangent_plane_coordinates). Then try to transfer from ENU to ECEF or Geo-coordiate. Hope this helps,
Best,
Welson,

Dinatorehanova, 2019-09-14:
Hi @welson

Can you please help me with the following queries how I can change from NED to ECEF? My imu data quaternions and position is in the NED frame how I can transfer into the ECEF frame ? I am also trying to geo reference my point cloud so that it overlaps with the real world place. I am stuck with it. It would be very helpful if could kindly refer any site or paper or a tutorial.

Thanks 

Dina

weisongwen, 2019-09-14:
Hi @Dinatorehanova , 
- to transfer the position (x,y,z) from NED/ENU to ECEF, please refer to this [function](https://github.com/weisongwen/amsi/blob/master/amsi/include/amsi/gnss_tools.hpp#L193). The inout is the reference point (longitude, latitude, altitude) and the current position in ENU as (x, y, z). The output is the ECEF position. 

- To sync the sensors, two methods are usually applied, the software-based and hardware-based. Based on your descriptions, the software-based method seems to be suitable. For the softwar-based method, you can simply find the closest timesatmps for each sensors or interpolate over the measurements. 

Hope these can help,

Best,
Weisong,

ethan20120, 2019-09-15:
@weisongwen ,

Thanks for the work. Is this code only for velodyne ? Can I use it for other LiDAR(e.g. Ouster)?

I have my odometry data in ENU frame.we already have bag file containing gps,imu,point cloud data.It seems your code performs the level arm offset, boresight matrix, rotation matrix by itself. How to feed the data into your code? Can you provide a demonstration?

Thanks 
Ethan

TixiaoShan, 2019-11-19:
Closed due to inactivity. Feel free to reopen if you have more questions.

closed 2019-11-19 by TixiaoShan

weisongwen, 2019-11-23:
@ethan20120  Sorry for my late reply. The function in my code is to transform the position from ENU to ECEF. if I am correct, what you want to do is to: transfer the IMU data (in NED) to the ECEF frame? Therefore, what you need to do is to rotate the IMU data using a rotation matrix which can be calculated based on your position (longitude, latitude, and altitude). The calculation of the matrix can be found in [here](http://web.stanford.edu/~gracegao/publications//journal//2019_Navigation_Probabilistic_Graphical_Fusion_of_LiDAR__GPS__and_3D_Building_Maps_for_Urban_UAV_Navigation_Derek%20Chen_final.pdf). please refer to the function (11). If you still have questions, you can send me your data, I can test it if I have time.:-)

Best,
Weisong,

ethan20120, 2019-11-29:
@weisongwen , Can you please use your function on this file to transform from the ENU to the ECEF frame? I am using a matlab function called enutogeodetic but the output file's point is not aligning with my GPS data.

The 6th, 7th, and the 8th column contains the east north up coordinate for the herewith attached file

Please find the herewith attached files for your kind review and consideration

https://drive.google.com/file/d/1PcahTNLhL3Z9XgOBhJcZaS4avdaKwqGQ/view?usp=sharing

I will wait for your kind reply.

Thanks 
Ethan

weisongwen, 2019-11-29:
@ethan20120  To transform the data from ENU to ECEF, we need a reference point (longitude, latitude, and altitude). See https://en.wikipedia.org/wiki/Local_tangent_plane_coordinates.

Best,
Weisong,

ethan20120, 2019-11-29:
@weisongwen , Sorry I forgot about it

40.25, -111.67, 1556.59

Its the reference point.

Can you please try

hunkyu, 2020-03-10:
@weisongwen  Have you succeeded in overlaying point clouds with google earth?