RobustFieldAutonomyLab/LeGO-LOAM #122: Point cloud coordinate offset

threeflowermeow, 2019-08-27:
Hi,
I used velodyne VLP16 Lidar to collect pcap data, converted it into a bagfile and entered the code. I found that the point cloud coordinate system displayed in rviz after calculation was deviated from the original coordinate system, which looked like the original Lidar coordinate system was rotated 90° counterclockwise. What causes it?
I found that atan2(x, y) was used to calculate horizonAngle in projectPointCloud of imageProjection.cpp, why not atan2(y, x)? It is related to the above question?
Thanks in advance.

closed 2019-08-29 by threeflowermeow

manomitbal, 2019-10-27:
Hi @threeflowermeow I am facing the same issue, were you able to find a solution to this? I transforming the final pointcloud the only step?

andre-nguyen, 2019-10-27:
The coordinate system from the VLP16 is different from the coordinate system used by the ROS driver [velodyne_driver](http://wiki.ros.org/velodyne_driver). The ROS driver has x pointing "forwards" i.e. away from the cable. Whereas the original Velodyne VLP16 coordinate frame has x pointing "right", 90 degrees from the cable.