RobustFieldAutonomyLab/LeGO-LOAM #124: LeGO-LOAM for all frames

jegor90, 2019-08-31:
Hello! I try to process all frames from sequence. Is it possible?  What should i change in code to process every frame?

ASajwan, 2019-09-05:
I don't understand your question completely. Do you mean KITTI's 'sequence'? If yes, convert KITTI's sequence into a bag file and feed it to LeGO_LOAM like they do with the examples in the repository's readme file and it will process the entire sequence.

jegor90, 2019-09-06:
Yes. I converted KITTI to rosbag. For example, there is 1000 frames (messages) with velodyne points. I print position (_odomAftMapped.pose.pose.position.x, odomAftMapped.pose.pose.position.y, odomAftMapped.pose.pose.position.z_) every time i generate it in _publishTF_ . As result, i have about 400 lines only instead of 1000. If i play .bag file with 
`  -r 0.05`
option, there is about 1200 lines in result file.
  
Maybe i need to change parameters?
```
extern const double mappingProcessInterval = 0.3;
extern const float scanPeriod = 0.1;
```

ASajwan, 2019-09-06:
Oh, so it is using time inside ROS and not the timestamps from the point cloud. May be changing some timing parameters would help. Sorry, I have no idea.

closed 2019-11-12 by stale[bot]