RobustFieldAutonomyLab/LeGO-LOAM #125: Evaluating on KITTI

jegor90, 2019-08-31:
I need to evaluate LeGO_LOAM on KITTI benchmark. There is a [tool](http://www.cvlibs.net/datasets/kitti/eval_odometry.php) on cvlibs for this. But how can i create a .txt file for this tool? Which numbers should i print there? I.e. we have (x, y, z) coordinate at each moment, how should i transform it into 12 numbers?

> Row i represents the i'th pose of the left camera coordinate system (i.e., z pointing forwards) via a 3x4 transformation matrix. The matrices are stored in row aligned order (the first entries correspond to the first row), and take a point in the i'th coordinate system and project it into the first (=0th) coordinate system.

ASajwan, 2019-09-05:
If you want to use LeGO_LOAM with KITTI, I think might be able to just convert KITTI to bag files using kitti2bag utility linked on KITTI's website and then use the appropriate ros topics and coordinate frames. Having said that, I haven't done this myself, but probably worth a try.

jegor90, 2019-09-06:
> If you want to use LeGO_LOAM with KITTI, I think might be able to just convert KITTI to bag files using kitti2bag utility linked on KITTI's website and then use the appropriate ros topics and coordinate frames. Having said that, I haven't done this myself, but probably worth a try.
Yes, i converted it. But i need .txt file for every sequence to compare it with ground truth (GT). Each line of GT contains of 12 numbers. To evaluate rotation error and translation error i must generate same file with LeGO_LOAM results.

ASajwan, 2019-09-06:
I see. I don't know which variables correspond to x,y,z and orientation of ego vehicle in LeGO-LOAM implementation. But once you have them, you can fix the first pose (x,y,z,3 angles) as the origin and then calculate the 3x4 transformation matrix based on that like inv(first Pose)*(current Pose). I am assuming that I know current pose here in world coordinates. If you know them only relative to the previous pose, just accumulate transformations over time by multiplying the transformation matrices. However, I think rather than comparing the 12 numbers of the matrix to the GT, probably you can compare translation and angles (expressed as Euler angles or quaternion) separately as it would tell you which parameter/direction is performing better and which is not.

closed 2019-11-12 by stale[bot]

bryantaoli, 2020-05-01:
if you test lego_loam on kitti, how do you set the parameter below:
// VLP-16
extern const int N_SCAN = 16;
extern const int Horizon_SCAN = 1800;
extern const float ang_res_x = 0.2;
extern const float ang_res_y = 2.0;
extern const float ang_bottom = 15.0+0.1;
extern const int groundScanInd = 7;

// HDL-32E
// extern const int N_SCAN = 32;
// extern const int Horizon_SCAN = 1800;
// extern const float ang_res_x = 360.0/float(Horizon_SCAN);
// extern const float ang_res_y = 41.33/float(N_SCAN-1);
// extern const float ang_bottom = 30.67;
// extern const int groundScanInd = 20;

// VLS-128
// extern const int N_SCAN = 128;
// extern const int Horizon_SCAN = 1800;
// extern const float ang_res_x = 0.2;
// extern const float ang_res_y = 0.3;
// extern const float ang_bottom = 25.0;
// extern const int groundScanInd = 10;