RobustFieldAutonomyLab/LeGO-LOAM #128: TransformToStart（featureAssociation）error

hehern, 2019-09-17:
when i run the lego_loam using my own bag,i crashed. I found that because the intensity of one point is nan,so after the conversion of TransformToStart, the coordinate(x,y,x) of the point is nan.
Then kdtreeSurfLast->nearestKSearch will crash.
So, 　I want to know what is the function of TransformToStart,and why needs intensity?
        float s = 10 * (pi->intensity - int(pi->intensity));
        float rx = s * transformCur[0];
        float ry = s * transformCur[1];
        float rz = s * transformCur[2];
        float tx = s * transformCur[3];
        float ty = s * transformCur[4];
        float tz = s * transformCur[5];

        float x1 = cos(rz) * (pi->x - tx) + sin(rz) * (pi->y - ty);
        float y1 = -sin(rz) * (pi->x - tx) + cos(rz) * (pi->y - ty);
        float z1 = (pi->z - tz);

        float x2 = x1;
        float y2 = cos(rx) * y1 + sin(rx) * z1;
        float z2 = -sin(rx) * y1 + cos(rx) * z1;

        po->x = cos(ry) * x2 - sin(ry) * z2;
        po->y = y2;
        po->z = sin(ry) * x2 + cos(ry) * z2;
        po->intensity = pi->intensity;

TixiaoShan, 2019-09-18:
The intensity here represents the time stamp of a point in the scan.

TixiaoShan, 2019-09-18:
This repo has some explanations for the code.

https://github.com/wykxwyc/LeGO-LOAM_NOTED

TixiaoShan, 2019-09-18:
You can also remove the nan point using pcl::removeNaNFromPointCloud()

closed 2019-09-19 by TixiaoShan

hehern, 2019-09-25:
> You can also remove the nan point using pcl::removeNaNFromPointCloud()

thanks