RobustFieldAutonomyLab/LeGO-LOAM #134: ouster lidar setting

louis960126, 2019-10-22:
Hi, 
I have an ouster 64 channel lidar. I already change the topic name to ouster lidar's topic name and change the utility.h to os1-64, but there is still some problem to run LeGO-LOAM. Do i need to change other settings?

Thank you!

louis960126, 2019-10-22:
For frame [base_link]: No transform to fixed frame [map]. TF error: [Lookup would require extrapolation into the past. Requested time 1464.161704880 but the earliest data is at time 1571715599.269487703, when looking up transform from frame [base_link] to frame [map]]

louis960126, 2019-10-22:
velodyne lidar is ok

louis960126, 2019-10-22:
[image]
[image]

TixiaoShan, 2019-10-22:
Try to comment this line.
https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/744254789c91ebe4851b4678be2c9f97e2f56c27/LeGO-LOAM/src/imageProjection.cpp#L159

closed 2019-10-28 by TixiaoShan

LeisureLei, 2019-12-23:
I meet the same problem.I play the bag file and I get nothing in rviz.There is alse a error in rviz :+1: For frame [base_link]: No transform to fixed frame [map]. TF error: [Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.]
I also uncomment line 159 and catkin_make agian.Nothin changes.
How can I fix this problem?

YiChenCityU, 2020-07-28:
> I meet the same problem.I play the bag file and I get nothing in rviz.There is alse a error in rviz  For frame [base_link]: No transform to fixed frame [map]. TF error: [Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.]
> I also uncomment line 159 and catkin_make agian.Nothin changes.
> How can I fix this problem?

Hi, have you solved this problem?

YiChenCityU, 2020-07-28:
> Hi,
> I have an ouster 64 channel lidar. I already change the topic name to ouster lidar's topic name and change the utility.h to os1-64, but there is still some problem to run LeGO-LOAM. Do i need to change other settings?
> 
> Thank you!

Hi, have you solved this problem?

Oyssster, 2021-03-07:
Hi,is there anyone solved this problem?

shahaizi0, 2022-06-30:
你好，请问有人解决了这个问题吗

orange-cr, 2023-04-29:
I run the park.bag provided by lio-sam and meet the sam problem. I solved this problem by:
(1)in run.launch set <param name="/use_sim_time" value="true" />
(2)rosbag play park.bag --clock
(3)uncomment this line in imageProjection.cpp at line 166
cloudHeader.stamp = ros::Time::now(); // Ouster lidar users may need to uncomment this line