RobustFieldAutonomyLab/LeGO-LOAM #136: i want to know why my rosbag is empty and how to translate my rosbag to a map

jokeY-1227, 2019-10-28:
i use "ros record -o OUT /laser_cloud_surround' to record the rosbag and i found that there's no data has been record, and i also use ' ros record -o OUT /velodyne_points' to record the data, and  is there anyway to translate the data record by the topic 'velodyne_points' to a map

TixiaoShan, 2019-10-28:
@jokeY-1227 
Try ```rosbag record /laser_cloud_surround``` to see if it works or not.
To transform a point cloud, you can use this [function](http://docs.ros.org/indigo/api/pcl_ros/html/namespacepcl__ros.html#aad20af010640a41f1898e75f6a6121ad).

jokeY-1227, 2019-10-31:
i have some questions with my step. first, " roslaunch the VLP16.yaml" ,then "rosrun rviz rviz -f velodyne",next, i try"rosbag record /laser_cloud_surround"and get no information from it ,just get a empty bag.
and i also try to launch legoloam and record the rosbag,but it warn me as "[ WARN] [1571916272.999377088]: /use_sim_time set to true and no clock published.  Still waiting for valid time..." 
what should i do to solve this?

TixiaoShan, 2019-10-31:
I wasn't able to run ```roslaunch the VLP16.yaml```. I will take a look how to let roslaunch take yaml file as command.

TixiaoShan, 2019-11-03:
Closed due to inactivity. Reopen if necessary.

closed 2019-11-03 by TixiaoShan