RobustFieldAutonomyLab/LeGO-LOAM #139: The point clouds are vertical after records as pcd files.

xingtongli, 2019-11-06:
HI,
I am working with Autoware and I want to use this algorithm to prepare the pcd map for using. I met a problem. I tried to use pointcloud_to_pcd to record and export the pointclouds as pcd files. After that, I import the pcd file into autoware. However I found that the pointclouds map is vertical. 
[image]
When I runing the algorithm, I saw the pointclouds are horizental in Rviz. Do you rotate the point clouds in the algorithm? I recorded the pointclouds from topic: laser_cloud_surround. What is the reason of this situation?

TixiaoShan, 2019-11-06:
Yes, the points are rotated here [https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/blob/744254789c91ebe4851b4678be2c9f97e2f56c27/LeGO-LOAM/src/featureAssociation.cpp#L487].

xingtongli, 2019-11-12:
Thank you for your reply, can you give me some hint of how to modify it? It does not work if I only modify the x,y,z.

TixiaoShan, 2019-11-16:
@xingtongli 
You need to rer-write all the code that is related to point cloud computation. It's probably not that easy.

closed 2019-11-19 by TixiaoShan