RobustFieldAutonomyLab/LeGO-LOAM #143: When publishing TF, how to understand the quaternion transformation?

heroacool, 2019-11-26:
[image]

Hi @TixiaoShan  , when reading the code, I'm confused with the quaternion transformation.
It seems that there is no abvious geometry meaning when convert a quaternion from (qx, qy, qz, qw) to (-qy, -qz, qx, qw). Would you please explain this operation?
Thanks!

TixiaoShan, 2019-11-26:
Ji Zhang likely converted odometry from the lidar frame to the camera frame for his later paper (V-LOAM). I think this is the reason behind this. Feel free to convert them back and publish odometry in the lidar frame for your own needs.

narutojxl, 2019-11-26:
Hi @TixiaoShan, Thank your work! According to your guess, i test as following, but the current laser pose in camera_0 frame do not consistent Ji Zhang's result, what do you think?  Look forward to your reply, thanks!

> 
	
   
        guess_test(transformSum[0], transformSum[1], transformSum[2]);
 
        void guess_test(float& sum0, float& sum1, float& sum2) {
		geometry_msgs::Quaternion q_camera_laser_msg, q_laserT_msg, result, geoQuat;
		tf::Quaternion q_camera_laser, q_laser_t;

		//camera_frame: X-Y-Z(right-down-front)
		//laser_frame:  X-Y-Z(left-up-front)

		q_camera_laser_msg = tf::createQuaternionMsgFromRollPitchYaw(0, 0, M_PI); //Rz(pi)Ry(0)Rx(0)
		// q_camera_laser_msg = tf::createQuaternionMsgFromRollPitchYaw(M_PI / 2.0, 0, M_PI / 2.0); //velodyne
		tf::quaternionMsgToTF(q_camera_laser_msg, q_camera_laser);

		//laser_t in laser_0 pose
		q_laserT_msg = tf::createQuaternionMsgFromRollPitchYaw(sum0, sum1, sum2); //Rz(sum2)Ry(sum1)Rx(sum0)
		tf::quaternionMsgToTF(q_laserT_msg, q_laser_t);

		//get laser_t in camera_0 pose
		q_camera_laser *= q_laser_t;
		tf::quaternionTFToMsg(q_camera_laser, result);
		std::cout << "We guess1:\n("
			      << result.x << " " << result.y << " " << result.z << " " << result.w << ")\n";
	

		geoQuat = tf::createQuaternionMsgFromRollPitchYaw(sum2, -sum0, -sum1);
		std::cout << "Zhang Ji:\n("
		          << -geoQuat.y << " " << -geoQuat.z << " " << geoQuat.x << " " << geoQuat.w << ")\n\n\n";
	    }

The result:

> We guess1:
> (-0.204232 0.305602 0.886847 -0.279998)
> Zhang Ji:
> (0.365856 -0.0354001 0.279998 0.886847)
> 
> 
> We guess1:
> (-0.184681 0.325077 0.88653 -0.272549)
> Zhang Ji:
> (0.370231 -0.0520693 0.272549 0.88653)

TixiaoShan, 2019-11-26:
Sorry, I don't understand what you are trying to do here - publish odometry in lidar frame?

narutojxl, 2019-11-26:
Sorry, i don't express myself clearly.  
As we know,  `transformSum` is the laser frame's pose in laser_0 frame(laser frame is X-Y-Z(left-up-front), laser_0 frame is the start position and attitude of laser frame ).

> Ji Zhang likely converted odometry from the lidar frame to the camera frame for his later paper (V-LOAM). I think this is the reason behind this.

I do a test to figure out what  rotation  transformation applied to  `transformSum's rotation` by Ji Zhang. The 3 transformations(red, blue, black) are all not equal to Ji Zhang published odometry TF,  you can see the  above result. What is the camera frame defined in your comment, something i missing?  Thanks for your help a lot :)
[image]

TixiaoShan, 2019-11-27:
Hello, the frame definition is from Ji Zhang's LOAM code, I didn't change anything of it.

TixiaoShan, 2019-11-27:
Another thread mentioned the frame conversion.
https://github.com/laboshinl/loam_velodyne/issues/134

heroacool, 2019-12-03:
> Ji Zhang likely converted odometry from the lidar frame to the camera frame for his later paper (V-LOAM). I think this is the reason behind this. Feel free to convert them back and publish odometry in the lidar frame for your own needs.

OK, Thanks

closed 2019-12-03 by heroacool