RobustFieldAutonomyLab/LeGO-LOAM #145: Problem about gtsam update & resize  in mapOptimization.cpp

heroacool, 2019-12-03:
Hi @TixiaoShan , when reading the mapOptimization.cpp, I am confusion with the gtsam update operation. What's the purpose of isam->update(), and why need to gtSAMgraph.resize(0) ?

[image]

TixiaoShan, 2019-12-03:
Hi @heroacool 
I think that's the way to use isam2 in GTSAM, it clears the values for next optimization. One example can be found here. https://github.com/devbharat/gtsam/blob/master/examples/VisualISAM2Example.cpp

heroacool, 2019-12-05:
> Hi @heroacool
> I think that's the way to use isam2 in GTSAM, it clears the values for next optimization. One example can be found here. https://github.com/devbharat/gtsam/blob/master/examples/VisualISAM2Example.cpp

Got it. Thanks !

closed 2019-12-05 by heroacool