RobustFieldAutonomyLab/LeGO-LOAM #146: Roslaunch Error

Yukina1996, 2019-12-16:
When I launch run.launch, it occurs to me:

setting /run_id to b046311a-1fd0-11ea-bdb0-e0d55eabd117
process[rosout-1]: started with pid [22553]
started core service [/rosout]
process[rviz-2]: started with pid [22562]
process[camera_init_to_map-3]: started with pid [22578]
process[base_link_to_camera-4]: started with pid [22579]
process[imageProjection-5]: started with pid [22589]
process[featureAssociation-6]: started with pid [22605]
process[mapOptmization-7]: started with pid [22618]
process[transformFusion-8]: started with pid [22638]
[ INFO] [1576479176.624368854]: ----> Image Projection Started.
[ INFO] [1576479176.630838361]: ----> Transform Fusion Started.
[ INFO] [1576479176.671602281]: ----> Map Optimization Started.
[featureAssociation-6] process has died [pid 22605, exit code -11, cmd /home/slam/SGGSLAM/WYY/lego_loam_ws/devel/lib/lego_loam/featureAssociation __name:=featureAssociation __log:=/home/slam/.ros/log/b046311a-1fd0-11ea-bdb0-e0d55eabd117/featureAssociation-6.log].
log file: /home/slam/.ros/log/b046311a-1fd0-11ea-bdb0-e0d55eabd117/featureAssociation-6*.log

I can run this program before, but suddenly this problem occurred.

TixiaoShan, 2019-12-16:
@Yukina1996 
At which part does the code crash?

Yukina1996, 2019-12-16:
> @Yukina1996
> At which part does the code crash?

When doing <roslaunch lego_loam run.launch>, it's wired that only the FeatureAssociation node is unable to established. Maybe there's a mistake in run.launch, because once it showed that syntax error near unexpected token `newline' in run.launch. But my run.launch is the same as yours.

Yukina1996, 2019-12-16:
> @Yukina1996
> At which part does the code crash?

When I doing catkin_make, I found that

CMake Warning at LeGO-LOAM/LeGO-LOAM/CMakeLists.txt:59 (add_executable):
  Cannot generate a safe linker search path for target transformFusion
  because files in some directories may conflict with libraries in implicit
  directories:

    link library [libpcl_common.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libpcl_kdtree.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libpcl_octree.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libpcl_search.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
    link library [libpcl_io.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

I don't know if this matters a lot?

TixiaoShan, 2019-12-17:
That's weird, I have never seen these problems before.

Yukina1996, 2019-12-18:
> That's weird, I have never seen these problems before.

THX for replying. The problem must be system environment. Some libs are conflicted, so I will trying reinstalling.

closed 2020-01-07 by TixiaoShan