RobustFieldAutonomyLab/LeGO-LOAM #150: If I increase the frequency of imu, what should I adjust in program?

carleft, 2019-12-24:
There is a const int imuQueLength in the program, and It has a value of 200. If the frequency of my imu is 400Hz, Should I change it to 400 or more? 

Thanks in advance!

TixiaoShan, 2019-12-24:
You don't need to change anything. The imu queue length is to store the imu data during one scan. So you only need to change it when your imu frequency is over 2000Hz. Just make sure ```imuQueLength > imu-frequency / lidar-frequency```

carleft, 2019-12-24:
> You don't need to change anything. The imu queue length is to store the imu data during one scan. So you only need to change it when your imu frequency is over 2000Hz. Just make sure `imuQueLength > imu-frequency / lidar-freque

OK , Thanks a lot~

closed 2019-12-24 by carleft