RobustFieldAutonomyLab/LeGO-LOAM #155: question about KDTree

zcg3648806, 2020-01-07:
When I run the official dataset downloaded from the home page url（https://github.com/RobustFieldAutonomyLab/jackal_dataset_20170608）,  I get this information:
///////////////
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
//////////////
What should I do?

TixiaoShan, 2020-01-07:
It means the input point cloud to kd-tree has zero points. Every scan in the sample dataset has enough features. So it's probably there is a bug in your modified code.

zcg3648806, 2020-01-08:
Thanks for your @TixiaoShan reply! As you said, I modified the parameters of the lidar（utlity.h), and when I change back to the default,the official dataset can work normally.

In addition, I find that there are almost eight times as many imu frames as there are radar frames,this means that in the official data set  the imu data frequency is very high, but in kitti dataset the imu has the same frequency as the radar. I think this is the main reason for the poor performance of the kitti dataset. what do you think？

TixiaoShan, 2020-01-08:
@zcg3648806 
The higher the IMU frequency the better. In fact, the IMU used in my dataset is very low-end and gives only 75Hz. It's recommended to use an IMU that is at least 200Hz, like the Xsens IMU used in Ji Zhang's paper.

zcg3648806, 2020-01-09:
Thanks for your @TixiaoShan reply! I got it.

closed 2020-01-10 by TixiaoShan