RobustFieldAutonomyLab/LeGO-LOAM #157: question about "void ShiftToStartIMU(float pointTime)"

zcg3648806, 2020-01-09:
This function is only defined, and I don't find it used anywhere else,Is this an obsolete function?

void ShiftToStartIMU(float pointTime)
    {
        imuShiftFromStartXCur = imuShiftXCur - imuShiftXStart - imuVeloXStart * pointTime;
        imuShiftFromStartYCur = imuShiftYCur - imuShiftYStart - imuVeloYStart * pointTime;
        imuShiftFromStartZCur = imuShiftZCur - imuShiftZStart - imuVeloZStart * pointTime;

        float x1 = cosImuYawStart * imuShiftFromStartXCur - sinImuYawStart * imuShiftFromStartZCur;
        float y1 = imuShiftFromStartYCur;
        float z1 = sinImuYawStart * imuShiftFromStartXCur + cosImuYawStart * imuShiftFromStartZCur;

        float x2 = x1;
        float y2 = cosImuPitchStart * y1 + sinImuPitchStart * z1;
        float z2 = -sinImuPitchStart * y1 + cosImuPitchStart * z1;

        imuShiftFromStartXCur = cosImuRollStart * x2 + sinImuRollStart * y2;
        imuShiftFromStartYCur = -sinImuRollStart * x2 + cosImuRollStart * y2;
        imuShiftFromStartZCur = z2;
    }

TixiaoShan, 2020-01-10:
It is used to deskew the point cloud. However, the translation estimation of IMU drifts super fast. So I observed bad performance when using this for deskewing.

You can certainly try it. See its place in the [code](http://docs.ros.org/indigo/api/loam_velodyne/html/scanRegistration_8cpp_source.html).

zcg3648806, 2020-01-10:
Thanks for your @TixiaoShan reply! I got it.

closed 2020-01-10 by TixiaoShan